CTSRD-CHERI/cheribsd #276: Stack-buffer overflowing write from getcwd() crashes the kernel

nwf, 2018-08-03:
Having made and entered a 4140-byte long path name, `/root/tmp/suzuki/suzuki/.../suzuki`, I ran a slightly modified bodiagsuite test `00183-large`, reproduced here, compiled as a MIPS executable.

```
#include <assert.h>
#include <errno.h>
#include <unistd.h>

int main(int argc, char *argv[])
{
  char buf[10];
  char * x;

  /*  BAD  */
  x = getcwd(buf, 4106);
  assert((x != NULL) || (errno == ERANGE));

  return 0;
}
```

I was greeted with this tale of sadness and woe

```
Trapframe Register Dump:
$0: 0                  at: 0                  v0: 0                  v1: 0x22              
a0: 0x4006a000         a1: 0x48               a2: 0x258bc0           a3: 0                 
a4: 0x1                a5: 0x7fffffb8a0       a6: 0x7fffffb898       a7: 0x1000            
t0: 0x66               t1: 0x6e               t2: 0x7fffffbd54       t3: 0x40321410        
s0: 0x7fffffcaf8       s1: 0x1                s2: 0x7fffffcae8       s3: 0                 
s4: 0                  s5: 0                  s6: 0                  s7: 0                 
t8: 0x1                t9: 0x400db870         k0: 0                  k1: 0                 
gp: 0x756b692f73757a75 sp: 0x7fffffca70       s8: 0x6b692f73757a756b ra: 0x692f73757a756b69
status: 0x408084b3 mullo: 0; mulhi: 0; badvaddr: 0x692f73757a756b69
cause: 0x48; pc: 0x692f73757a756b69
panic: pcc (1757a756b69) <-> pc (692f73757a756b69) mismatch:
time = 1533331591
KDB: enter: panic
[ thread pid 1438 tid 100057 ]
Stopped at      kdb_enter+0x94: break   0
db> bt
Tracing pid 1438 tid 100057 td 0x9800000001e8b000
kdb_enter+0x94 (?,?,?,?) ra ffffffff803e3614 sp c0000000003f7810 sz 32
vpanic+0x1c4 (?,?,?,?) ra ffffffff803e344c sp c0000000003f7830 sz 64
kassert_panic+0x254 (?,?,?,?) ra 0 sp c0000000003f7870 sz 0
```

That pc seems to be "i/suzuki", which is really only to be expected, so I surely do not expect the program to continue running for long, but the kernel panic is unfortunate.

brooksdavis, 2018-08-03:
Hmm.  Glancing at the implementation, __getcwd() is limited to 1024 bytes so I suspect a bug in `vn_fullpath()` for long paths.  It's probably worth trying to reproduce in an x86 vm.

nwf, 2018-08-03:
Well, in the test, there's certainly less than 1024 bytes between the buffer and at least one stored instruction pointer, so that particular limit may not matter.  I was only using the huge path because I saw "4K" and didn't think too hard about it. ;)

brooksdavis, 2018-08-03:
Yeah, that makes sense.  Ah, the panic is in fetch_instr_near_pc().  I think the assertion is just plain wrong.

arichardson, 2018-08-04:
Yeah we can remove that assertion although I would expect the epc to be equal to  getoffset($epcc) whenever we enter a trap.

The assertion helped me track down a bug in the fpga where EPCC points to the branch delay slot but EPC to the branch.

arichardson, 2018-08-04:
I guess changing the assertion to be getaddr($pcc) == ($pc) would work?

arichardson, 2018-08-04:
Possibly the reason that EPC and EPCC are not in sync is that we are not actually getting a trap caused by an instruction but actually are failing in instruction fetch. I think `log_frame()` always tries to fetch the bad instruction, but in this case it is nonsense.
I still wonder why we are ending up with an EPC that is not EPCC.offset.

brooksdavis, 2018-08-14:
It looks like @arichardson fixed this

closed 2018-08-14 by brooksdavis