CTSRD-CHERI/cheribsd #285: Kernel panic when executing ld-cheri-elf.so.1 directly

arichardson, 2018-08-31:
```
$ /libexec/ld-cheri-elf.so.1 --help
panic: Constructed capability has wrong length 0x7ffff00000 != 0x7fffee0000
time = 1535708110
KDB: enter: panic
[ thread pid 657 tid 100043 ]
Stopped at      kdb_enter+0x94: break   0
db> bt
Tracing pid 657 tid 100043 td 0x98000000022b0000
kdb_enter+0x94 (?,?,?,?) ra ffffffff80422c6c sp c0000000002b73a0 sz 32
vpanic+0x224 (?,?,?,?) ra ffffffff804228d8 sp c0000000002b73c0 sz 64
doadump (?,?,?,?) ra 0 sp c0000000002b7400 sz 0
db>
```

arichardson, 2018-10-25:
Just ran into this again:
```
env LD_CHERI_TRACE_LOADED_OBJECTS=yes /rootfs/libexec/ld-cheri-elf.so.1 /usr/libcheri/libctf.so
panic: Constructed capability has wrong length 0x7ffff00000 != 0x7fffee0000
```

Should we just print a warning and refuse to load the ELF file instead? At least that way the system stays alive.

rwatson, 2018-10-25:
A KASSERT is appropriate if the operation should never fail, as it would reflect a violation of a kernel invariant. Presumably we have a bug .. is this on CHERI-128 or CHERI-256?

arichardson, 2018-11-04:
This has been fixed in https://github.com/CTSRD-CHERI/cheribsd/commit/bea45e4866cb38fe42cced7b839f0eb909150905, https://github.com/CTSRD-CHERI/cheribsd/commit/5514f9943ad270bc1f39ba85f4dc32a46fdd08ac, https://github.com/CTSRD-CHERI/cheribsd/commit/d4b5e59d12ad20054547f19f1bcc72bff7109166, https://github.com/CTSRD-CHERI/cheribsd/commit/1cd9ba4c6fbc4553add6973889c61fac65d65c81 and https://github.com/CTSRD-CHERI/cheribsd/commit/50755a9fcebeb41b6c96f566592c8b899fb9be90

closed 2018-11-04 by arichardson