CTSRD-CHERI/cheribsd #287: Unknown instructions in userspace branch delay slot cause kernel panic

arichardson, 2018-11-04:
While writing some userspace assembly code I accidentally forgot to place a nop (I had a `teq $zero, $zero there) in the $cjr delay slot.
This caused the kernel to panic instead of my userspace program being killed.

```
KDB: enter: BAD OPCODE in MipsEmulateBranch
```

We should probably print an error message and abort the user program instead of killing the whole system.

brooksdavis, 2020-04-21:
It seems a certain @arichardson person added this drop to DDB in 6565e4c8f853e111bfde70497cac83f1ad1b4c56. 😄 We might just want to revert that change.

arichardson, 2020-04-21:
Three months is a long time... But I think we still don't handle CJR/CJALR in this code.

bsdjhb, 2022-02-09:
OBE as MIPS is now gone.

closed 2022-02-09 by bsdjhb