CTSRD-CHERI/cheribsd #323: SIGALARM caused by timer_create() doesn't set sival_ptr

arichardson, 2019-08-04:
Reproducer: `/usr/tests/cheri/lib/libc/sys/timer_create_test -vunprivileged-user=tests -r/tmp/kyua.1NlSsY/1316/result.atf timer_create_mono`
This crashes when trying to dereference si->si_value.sival_ptr.
```
Program received signal SIGPROT, CHERI protection violation
Capability tag fault caused by register c1: 0x0000000000000000 [rwRW,0x0-0xffffffffffffffff].
0x0000000000131594 in timer_signal_handler (signo=14, si=0x7ffffce520 [rwRW,0x7ffffce520-0x7ffffce590],
    osi=0x7ffffce010 [rwRW,0x7ffffce010-0x7ffffce520])
    at /local/scratch/alr48/cheri/cheribsd/contrib/netbsd-tests/lib/libc/sys/t_timer_create.c:47
```
GDB tells me that the sival_ptr value contains nonsense rather than the timer pointer:
```
(gdb) p/x *si
$4 = {si_signo = 0xe, si_errno = 0x0, si_code = 0x10003, si_pid = 0x0, si_uid = 0x0, si_status = 0x0,
  si_addr = 0x4037ef60, si_value = {sival_int = 0xf17d0000, sival_ptr = 0x0, sigval_int = 0xf17d0000,
    sigval_ptr = 0x0}, _reason = {_fault = {_trapno = 0x0, _capreg = 0x0}, _timer = {_timerid = 0x0,
      _overrun = 0x0}, _mesgq = {_mqd = 0x0}, _poll = {_band = 0x0}, __spare__ = {__spare1__ = 0x0, __spare2__ = {
        0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0}}}}
```

brooksdavis, 2019-08-05:
Part of the problem is fixed in 4b7e29ca71fa25d282cf0e4665877025ad487f25.  If that is applied to the caprevoke branch then the test passes.  On master it just fails differently.

brooksdavis, 2020-04-21:
Seems to be fixed:
```
root@qemu-cheri128-bed22:~ # file /usr/tests//lib/libc/sys/timer_create_test
/usr/tests//lib/libc/sys/timer_create_test: ELF 64-bit MSB shared object, MIPS, 
MIPS-IV with CHERI-128 (CheriABI) version 1 (FreeBSD), dynamically linked, interpreter /libexec/ld-elf.so.1, for FreeBSD 13.0 (1300092), FreeBSD-style, with deb
ug_info, not stripped
root@qemu-cheri128-bed22:~ # /usr/tests//lib/libc/sys/timer_create_test -vunprivileged-user=tests timer_create_mono
timer_create_test: WARNING: Running test cases outside of kyua(1) is unsupported
timer_create_test: WARNING: No isolation nor timeout control is being applied; y
ou may get unexpected failures; see atf-test-case(4)
timer_signal_handler: Alarm clock
passed
```

closed 2020-04-21 by brooksdavis