CTSRD-CHERI/cheribsd #331: time -l reports 0

jonwoodruff, 2019-09-19:
/usr/bin/time -l gives:

      102.76 real       100.43 user         2.23 sys
         0  maximum resident set size
         0  average shared memory size
         0  average unshared data size
         0  average unshared stack size
         0  page reclaims
         0  page faults
         0  swaps
         0  block input operations
         0  block output operations
         0  messages sent
         0  messages received
         0  signals received
         0  voluntary context switches
         0  involuntary context switches

brooksdavis, 2019-09-20:
What does `file /usr/bin/time` say?  A CheriABI binary works for me:

```
root@beri1:~ # file /usr/bin/time       
/usr/bin/time: ELF 64-bit MSB shared object, MIPS, MIPS-IV with CHERI-128 (Cheri
ABI) version 1 (FreeBSD), dynamically linked, interpreter /libexec/ld-cheri-elf.
so.1, for FreeBSD 13.0 (1300036), FreeBSD-style, not stripped
root@beri1:~ # /usr/bin/time -l echo foo
foo
        0.14 real         0.05 user         0.07 sys
      2324  maximum resident set size
         4  average shared memory size
         4  average unshared data size
       256  average unshared stack size
       139  page reclaims
         0  page faults
         0  swaps
         0  block input operations
         0  block output operations
         0  messages sent
         0  messages received
         0  signals received
         2  voluntary context switches
         6  involuntary context switches
```

brooksdavis, 2019-09-20:
Hmm, also works for me with mips:
```
root@beri1:~ # file ~ctsrd/time 
/home/ctsrd/time: ELF 64-bit MSB shared object, MIPS, MIPS-IV version 1 (FreeBSD
), dynamically linked, interpreter /libexec/ld-elf.so.1, for FreeBSD 13.0 (13000
36), FreeBSD-style, not stripped
root@beri1:~ # ~ctsrd/time -l echo foo
foo
        0.13 real         0.05 user         0.06 sys
      2328  maximum resident set size
         4  average shared memory size
         4  average unshared data size
       256  average unshared stack size
       104  page reclaims
         0  page faults
         0  swaps
         0  block input operations
         0  block output operations
         0  messages sent
         0  messages received
         0  signals received
         2  voluntary context switches
         6  involuntary context switches
```

Both cases are qemu.

arichardson, 2019-09-20:
@jonwoodruff I wonder if your rootfs and kernel are out-of-sync?

brooksdavis, 2020-04-21:
Please reopen if this is reproducible

closed 2020-04-21 by brooksdavis