CTSRD-CHERI/cheribsd #342: Malicious malloc consumer can violate spatial safety

brettferdosi, 2019-11-27:
If a malicious consumer of jemalloc makes an allocation, increments the offset, then frees the capability, the next allocation will preserve the consumer-set offset while giving them the "full" length, which allows the consumer to access memory belonging to the next allocation.

`
#include <stdlib.h>
#include <stdio.h>
#include <malloc_np.h>
#include <cheri/cheric.h>

int main() {
        char *myptr = (char *)malloc(16);
        printf("myptr %#p\n", myptr);
        myptr += 3;
        free(myptr);
        char *newalloc = (char *)malloc(16);
        printf("newalloc 1 %#p\n", newalloc);
        newalloc = (char *)malloc(16);
        printf("newalloc 2 %#p\n", newalloc);
}
`

Running the above code gives:

`
myptr v:1 s:0 p:0006817d b:0000000040862000 l:0000000000000010 o:0 t:-1
newalloc 1 v:1 s:0 p:0006817d b:0000000040862003 l:0000000000000010 o:0 t:-1
newalloc 2 v:1 s:0 p:0006817d b:0000000040862010 l:0000000000000010 o:0 t:-1
`

brooksdavis, 2020-04-30:
The sanity checks in 00cae89b21b04489fadf7658150a31d0abfdfb2c resolve this issue for an attacker who can't set bounds. With subobject bounds enabled code like this may still be a problem:
```C
struct s {
  int foo;
  int bar;
};

int f() {
  struct s *an_s;
  void *newptr;
  an_s = malloc(sizeof(*an_s));
  free(&an_s->bar);
  newptr = malloc(sizeof(*an_s));
}
```
It's likely possible to add further checking against jemalloc metadata to catch some cases where the resulting base is absurdly aligned.

trasz, 2020-07-13:
Looking into it.