CTSRD-CHERI/cheribsd #353: __cheri_addr with dependent type '<dependent type>' is not yet supported

OCTAGRAM, 2020-01-07:
Environment: Debian Stretch in WSL. Building world with:

`wsl -d Debian ./cheribuild.py --cheri-256 --qemu/no-use-smbd all`

Getting 6 similar errors:

`
>>> stage 4.2.1: building CHERI libraries
===> lib/libgcc_eh (obj,all,install)
In file included from /home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/libunwind.cpp:22:
In file included from /home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/UnwindCursor.hpp:69:
/home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/DwarfInstructions.hpp:113:45: error: __cheri_addr with dependent type '<dependent type>' is not yet supported
                                            _pint_to_addr(savedReg.value));
                                            ^
/home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/AddressSpace.hpp:397:30: note: expanded from macro '_pint_to_addr'
#define _pint_to_addr(val) ((__cheri_addr LocalAddressSpace::addr_t)val)
                             ^
In file included from /home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/libunwind.cpp:22:
In file included from /home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/UnwindCursor.hpp:69:
/home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/DwarfInstructions.hpp:140:45: error: __cheri_addr with dependent type '<dependent type>' is not yet supported
    return addressSpace.getCapability(cfa + _pint_to_addr(savedReg.value));
                                            ^
/home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/AddressSpace.hpp:397:30: note: expanded from macro '_pint_to_addr'
#define _pint_to_addr(val) ((__cheri_addr LocalAddressSpace::addr_t)val)
                             ^
In file included from /home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/libunwind.cpp:22:
In file included from /home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/UnwindCursor.hpp:69:
/home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/DwarfInstructions.hpp:171:41: error: __cheri_addr with dependent type '<dependent type>' is not yet supported
    return addressSpace.getDouble(cfa + _pint_to_addr(savedReg.value));
                                        ^
/home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/AddressSpace.hpp:397:30: note: expanded from macro '_pint_to_addr'
#define _pint_to_addr(val) ((__cheri_addr LocalAddressSpace::addr_t)val)
                             ^
In file included from /home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/libunwind.cpp:22:
In file included from /home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/UnwindCursor.hpp:69:
/home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/DwarfInstructions.hpp:194:41: error: __cheri_addr with dependent type '<dependent type>' is not yet supported
    return addressSpace.getVector(cfa + _pint_to_addr(savedReg.value));
                                        ^
/home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/AddressSpace.hpp:397:30: note: expanded from macro '_pint_to_addr'
#define _pint_to_addr(val) ((__cheri_addr LocalAddressSpace::addr_t)val)
                             ^
In file included from /home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/libunwind.cpp:22:
In file included from /home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/UnwindCursor.hpp:69:
/home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/DwarfInstructions.hpp:593:22: error: __cheri_addr with dependent type 'libunwind::DwarfInstructions::sint_t' (aka 'typename A::sint_t') is not yet supported
      *sp = (pint_t)(_pint_to_addr(svalue2) * _pint_to_addr(svalue));
                     ^
/home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/AddressSpace.hpp:397:30: note: expanded from macro '_pint_to_addr'
#define _pint_to_addr(val) ((__cheri_addr LocalAddressSpace::addr_t)val)
                             ^
In file included from /home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/libunwind.cpp:22:
In file included from /home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/UnwindCursor.hpp:69:
/home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/DwarfInstructions.hpp:593:47: error: __cheri_addr with dependent type 'libunwind::DwarfInstructions::sint_t' (aka 'typename A::sint_t') is not yet supported
      *sp = (pint_t)(_pint_to_addr(svalue2) * _pint_to_addr(svalue));
                                              ^
/home/octagram/cheri/cheribsd/contrib/subrepo-cheri-libunwind/src/AddressSpace.hpp:397:30: note: expanded from macro '_pint_to_addr'
#define _pint_to_addr(val) ((__cheri_addr LocalAddressSpace::addr_t)val)
                             ^
6 errors generated.
--- libunwind.o ---
`

Can't get through. Is there downloadable disk of CheriBSD?

arichardson, 2020-01-07:
Sorry about that. You will have to update LLVM.

OCTAGRAM, 2020-01-07:
When I use cheribuild.py, what does it take to update LLVM?

arichardson, 2020-01-07:
It should be automatic (unless you pass `--skip-update`). I realized I hadn't merged the pull request with the fix yet. If you build now it should work (after running `cheribuild.py llvm`).

OCTAGRAM, 2020-01-07:
Running into another errors now. This one is gone. Thanks.

closed 2020-01-08 by arichardson