CTSRD-CHERI/cheribsd #354: tcpdump-helper is referencing discarded DLT_IPMB symbol

OCTAGRAM, 2020-01-08:
Environment: Debian Stretch in WSL. Command line:
`wsl -d Debian ./cheribuild.py --cheri-256 --qemu/no-use-smbd all`

Error:

`
>>> stage 4.4: building everything
/home/octagram/cheri/cheribsd/libexec/tcpdump-helper/pcap_datalink_val_to_name.c:127:13: error: use of undeclared identifier 'DLT_IPMB'
        DLT_CHOICE(DLT_IPMB, "IPMB"),
                   ^
1 error generated.
--- pcap_datalink_val_to_name.o ---
*** [pcap_datalink_val_to_name.o] Error code 1
bmake[4]: stopped in /home/octagram/cheri/cheribsd/libexec/tcpdump-helper
1 error
`

Comments in dlt.h hint me that this line can be safely removed. That fixed building userspace.

arichardson, 2020-01-08:
I'm not sure we need to build this directory anymore.
A workaround should be to delete line 32 in libexec/Makefile (`${_tcpdump_helper}`).

I'll defer to @brooksdavis for the correct solution.

brooksdavis, 2020-01-08:
Fixed with 37a94d9178ba822b987a5a54f761b2522639a23f

closed 2020-01-08 by brooksdavis