CTSRD-CHERI/cheribsd #377: Various kernel modules fail to build

arichardson, 2020-01-31:
For example:

`/local/scratch/alr48/cheri/cheribsd/sys/dev/mfi/mfi.c:3568:20: error: implicit declaration of function 'PTRIN' is invalid in C99 [-Werror,-Wimplicit-function-declaration]`

I will work around this with NO_MODULES for now but it would probably be nice to fix.

bsdjhb, 2020-01-31:
I found that these were usually due to missing "freebsd32.h" or "freebsd64.h" includes.  I thought I had pushed fixes for all of these though as I ran into it building risc-v already.

bsdjhb, 2020-01-31:
#367 included some fixes for this file after the changes to remove iovec shims.  Which kernel config is failing?  Does it only have COMPAT_CHERIABI and not COMPAT_FREEBSD64 perhaps?

arichardson, 2020-01-31:
This is GENERIC and QEMU without any Cheri support.

brooksdavis, 2020-04-21:
I did a quick test build and it looks like these are mostly issues that @bsdjhb's EXPLICIT_USER_ACCESS branch should fix so I'll assign to him for now.  Happy to jump in to help if there are other issues.

brooksdavis, 2020-05-27:
I think this should be OBE?

bsdjhb, 2020-05-27:
This should be fixed. I built both QEMU and CHERI_QEMU for testing during the explicit work.

closed 2020-05-27 by bsdjhb