CTSRD-CHERI/cheribsd #393: jefree doesn't check that the capability is valid before freeing it

nicolasjms, 2020-02-24:
jefree doesn't check that the capability is valid before freeing it. An attacker can abuse this combined with issue 342 to forge an arbitrary capability. Example:
myptr (before corruption): 0x408c9000 - v:1 s:0 p:0006817d b:00000000408c9000 l:0000000000000400 o:0 t:-1
myptr (after corruption) : 0x408c9008 - v:0 s:0 p:0006817d b:00000000408c9000 l:0000000000000400 o:8 t:-1
myptr2:                    0x408c9008 - v:1 s:0 p:0006817d b:00000000408c9008 l:0000000000000400 o:0 t:-1

int main() {
    void *myptr = (void **)malloc(0x400);
    
    printf("myptr (before corruption): %p - %#p\n", myptr, myptr);
     
    char *data = (char *)&myptr;
    data[0xf] = 8;
    
    printf("myptr (after corruption) : %p - %#p\n", myptr, myptr);
    
    free(myptr);
    
    void *myptr2 = malloc(0x400);
    
    printf("myptr2:                    %p - %#p\n", myptr2, myptr2);
    
}

nicolasjms, 2020-02-24:
This also works with realloc btw.

```
int main() {
    void *myptr = (void **)malloc(0x400);
    
    printf("myptr (before corruption): %p - %#p\n", myptr, myptr);
     
    char *data = (char *)&myptr;
    data[0xf] = 8;
    
    printf("myptr (after corruption) : %p - %#p\n", myptr, myptr);
    
    //free(myptr);
    myptr = realloc(myptr, 0x408);
    
    void *myptr2 = malloc(0x400);
    
    printf("myptr:                    %p - %#p\n", myptr, myptr);
    printf("myptr2:                    %p - %#p\n", myptr2, myptr2);
	
    return 0;
```

outputs:

```
myptr (before corruption): 0x408c9000 - v:1 s:0 p:0006817d b:00000000408c9000 l:0000000000000400 o:0 t:-1
myptr (after corruption) : 0x408c9008 - v:0 s:0 p:0006817d b:00000000408c9000 l:0000000000000400 o:8 t:-1
myptr:                    0x408e3000 - v:1 s:0 p:0006817d b:00000000408e3000 l:0000000000000408 o:0 t:-1
myptr2:                    0x408c9008 - v:1 s:0 p:0006817d b:00000000408c9008 l:0000000000000400 o:0 t:-1
```

rwatson, 2020-02-24:
Thanks for this report!

Tagging @brooksdavis and @nwf.

While the above isn’t exactly within the threat model for the baseline CheriBSD, it’s clearly a bad thing and should be fixed. (For clarity, only, this doesn’t allow the forging of arbitrary capabilities, only those that the memory allocator was itself able to derive. This is an academic distinction in the current branch, other than that excluding forging of kernel capabilities, but a very important one for our upcoming work on process separation using CHERI within a single address space.)

@nwf can point you at our branch on temporal and general allocator memory safety, which enforces much stronger protections around memory allocation. This is not yet in the CheriBSD mainline, but will hopefully be merged there in the next few months. It would be good to know if this problem is resolved there, but also, if you are interested in these aspects of allocator hardening, you will be interested in looking in that branch in more detail.

rwatson, 2020-02-27:
And @brooks: Regardless of this being within the specific threat model of jemalloc in the main branch, it’s clearly not a preferred behaviour and is likely worth eliminating?

davidchisnall, 2020-02-27:
Snmalloc includes checks that would prevent this kind of attack, so replacing jemalloc with a snmalloc would close this vulnerability.

chertl, 2020-02-27:
Root caused to this `CSetAddr` instruction:

https://github.com/CTSRD-CHERI/cheribsd/blob/f3bd7facb07f6baad5a16397ef811474be2cf18c/contrib/jemalloc/include/jemalloc/internal/jemalloc_internal_inlines_c.h#L38

The capability `ptr` is casted into an integer for the instruction, losing the tag validity information.

nwf, 2020-02-27:
@brettferdosi Is the resident expert in this particular detail, but I believe the situation is roughly as follows.  We have thought quite a bit about the requirements on allocators to prevent their use as amplification oracles and to ensure the correctness of their heap data structures in the face of adversarial inputs.

- Our https://github.com/CTSRD-CHERI/mrs wrapper imposes a requirement of a new API, `malloc_underlying_allocation`, that the wrapper can use to perform most of this validation external to the allocator itself.  Things are additionally tricky when we need to ensure that the temporal-safety quarantine mechanism does not itself introduce double-frees, but that is somewhat tangential.

- Our https://github.com/CTSRD-CHERI/dlmalloc_nonreuse patches to `dlmalloc` include hardening to ensure that the capabilities being given to `free` are exactly those as issued: after it has found the purported arena-wide capability, it ensures that the user-provided capability is a copy of one stored prior to its base (i.e., before the allocated chunk and not subject to tampering by the user program).  This filters out adversarial inputs to `free`, but is fairly expensive in terms of heap storage.

- Our https://github.com/CTSRD-CHERI/snmalloc CHERI-fication effort includes (or should include... there may have been some violence with all the branches floating around) checks that the capability is tagged and has offset 0 (and has non-zero permission word?  I don't recall if that made it in, but it should); `snmalloc` itself already contains checks that the purported object being freed is suitably aligned within the arena containing it.  The slab-like nature of `snmalloc`'s allocation arenas means that we do not need to store additional data to perform this test, but it does complicate the `free` pass.  In combination with the `mrs` wrapper, we defer `free`s and pay the increased cost of `free` itself on a different thread.

- `jemalloc` is pretty complicated internally, and even the `malloc_underlying_allocation` function we have for it is approximate and so it remains vulnerable to corrupting its data structures and mis-issuing future capabilities, including use as an amplification oracle.  (`jemalloc` is in fact pretty ideal for this, since its per-thread magazines mean it's pretty likely that you get back the thing you just freed, or, rather, `jemalloc`'s idea of what you just freed...)  I suspect we will not fix this due to the large engineering effort required, but this means we should transition away from `jemalloc` as the default allocator, probably to a hardened `snmalloc`.  @davidchisnall has provided (but I have not yet integrated) patches to cheribsd to make `snmalloc` part of its build; the upstreaming of CHERI-ficiation of `snmalloc` is proceeding in fits and spurts as I have time between other things but is currently stalled and needs some refactoring work.

Sorry that's long-winded.  If I've said anything incorrect, I'm sure someone will chime in. :)

nicolasjms, 2020-02-28:
Another thing to keep in mind when using jemalloc, it does not protect against double frees. Freeing twice the same pointer will result in having it stored twice in the cache, which can later be abused to cause type confusions. From my understanding snmalloc does not have this issue.

brettferdosi, 2020-02-28:
@nwf 's characterization above is correct. `snmalloc` and `dlmalloc` are robust to malformed input to free (our snmalloc uses existing machinery to detect #342 augmented with some CHERI-specific checks, and dlmalloc uses a stored capability to validate the passed-in one). `jemalloc` has not yet been fully hardened. `jemalloc`'s rather odd double-free behavior would be fixed with a simple validity check when using our `mrs` wrapper for temporal safety, since  the wrapper would either detect and prevent the double-free when painting the revocation shadow bitmap, or the second free attempt would be with an invalid capability.

nwf, 2020-02-28:
It's important to emphasize that our CHERI-fied `snmalloc` is only really robust against double-free when we're doing quarantining/temporal safety, with either the earlier native integration of revocation or the `mrs` wrapper; its native double-free detection relies on in-band data, which is not, in general, safe from adversarial tampering.  This holds of the upstream, non-CHERI `snmalloc` as well, I think.

`dlmalloc`'s expensive "store a copy of the capability out of band" game lets it clear that capability in a way that can't be restored by an adversary and so it can catch double-frees precisely.  By doubling the out-of-band data to *two* capabilities, it can additionally be made robust against adversarial attempts to manipulate its free list (by moving the metadata out to the header), but this is getting even more expensive.  I believe that our version does both, but @brooksdavis is the expert there.

brooksdavis, 2020-02-28:
`dlmalloc` does indeed expand metadata to two pointers.  This allows us to consolidate allocations in quarantine and reduce list size (as well as unmapping more of the interior of large regions that are free'd together).  As a side effect, it leaves us enough space to store a copy of the allocation which the can check in the free path for sanity and also deal with attempts to free wrong-sized objects.

I've made not attempt to figure out if we can make `jemalloc` more robust. We should discuss this with the authors at some point.

brooksdavis, 2020-04-21:
The short-term plan of action is to switch to snmalloc as the default CheriABI allocator to address this issue (and because we've had more success understanding it's internals for our temporal safety work.)

brooksdavis, 2020-04-21:
Just realized I'd mixed up which jemalloc issue this was.  I'm currently looking at a fix for this issue.

brooksdavis, 2020-04-30:
Other than merging to master I believe this issue is resolved.

closed 2020-04-30 by brooksdavis