CTSRD-CHERI/cheribsd #406: Unexpected cheriabitest failures

nwf, 2020-03-23:
Running 365e480c373c9b3d72b92dda8e1bf7848a83e6e9, `cheriabitest` reports:

```
Unexpected failures:
  test_initregs_stack: stack offset 3fdcf50 (expected range 7fc000 to 7fc000)
  test_bounds_global_static_uint8_array65537: global: incorrect length (expected 65537, got 65664)
  test_bounds_global_uint8_array65537: global: incorrect length (expected 65537, got 65664)
  test_bounds_extern_global_array65537: global: incorrect length (expected 65537, got 65664)
  test_unaligned_capability_copy_memcpy: Expected child termination with signal 6
  test_unaligned_capability_copy_memmove: Expected child termination with signal 6
```

I think these result from...
* `initregs_stack` : having a larger stack than when the test was written
* `bounds_*`,  stronger Concentrate alignment requirements than when the tests were written
* `unaligned_*`, `memcpy`/`memmove` now just strips the tag and doesn't abort in these cases

If that's right (@arichardson is probably the person who knows), I'll go ahead and update the tests in the obvious ways.

arichardson, 2020-03-23:
Yes, all caused by changing stuff without updating tests. Thanks!

jrtc27, 2020-03-23:
I recognise all those as the usual known failures I've seen and ignored for the past ~forever :)

closed 2020-04-06 by arichardson