CTSRD-CHERI/cheribsd #413: `sched_switch` NULL dereference panic?

nwf, 2020-03-28:
This might be quite stale, since it's using the Cornucopia kernel builds, but just in case:

```
root@qemu-test:~ # /usr/sbin/devctl disable atse0
KDB: stack backtrace:
db_trace_self+0x18 (?,?,?,?) ra ffffffff8015ee08 sp c000000028246850 sz 16
db_fetch_ksymtab+0x240 (?,?,?,?) ra ffffffff803d11f4 sp c000000028246860 sz 800
kdb_backtrace+0x74 (?,?,?,?) ra ffffffff8064cd48 sp c000000028246b80 sz 16
vm_fault+0x2840 (?,?,?,?) ra ffffffff8064a390 sp c000000028246b90 sz 464
vm_fault_trap+0x28 (?,?,?,?) ra ffffffff806b48c4 sp c000000028246d60 sz 32
trap+0xbac (?,?,?,?) ra ffffffff8069c6ec sp c000000028246d80 sz 272
MipsKernGenException+0x1e0 (?,?,?,?) ra 0 sp c000000028246e90 sz 0
--- exception, cause 9800000006c8d000 badvaddr 0 ---
sched_switch+0xb70 (?,?,?,?) ra ffffffff803717ac sp c000000028247210 sz 160
mi_switch+0x1bc (?,?,?,?) ra ffffffff803aea04 sp c0000000282472b0 sz 48
sched_bind+0x9c (?,?,?,?) ra ffffffff803c7e6c sp c0000000282472e0 sz 32
epoch_drain_callbacks+0x2a4 (?,?,?,?) ra ffffffff804e09a0 sp c000000028247300 sz 160
if_detach+0x2d0 (?,?,?,?) ra ffffffff804efa2c sp c0000000282473a0 sz 112
ether_ifdetach+0x64 (?,?,?,?) ra ffffffff806cccb0 sp c000000028247410 sz 32
atse_attach+0x2a58 (?,?,?,?) ra ffffffff806ccde0 sp c000000028247430 sz 80
atse_detach_dev+0x18 (?,?,?,?) ra ffffffff803b5734 sp c000000028247480 sz 32
device_detach+0x2dc (?,?,?,?) ra ffffffff803c0edc sp c0000000282474a0 sz 96
_gone_in_dev+0x25c4 (?,?,?,?) ra ffffffff8021b0ec sp c000000028247500 sz 224
dev2udev+0x3dec (?,?,?,?) ra ffffffff806dc7a8 sp c0000000282475e0 sz 96
VOP_IOCTL_APV+0x20 (?,?,?,?) ra ffffffff804b8e74 sp c000000028247640 sz 16
user_copy_file_range+0x8ac (?,?,?,?) ra ffffffff8021bb1c sp c000000028247650 sz 288
dev2udev+0x481c (?,?,?,?) ra ffffffff80401b0c sp c000000028247770 sz 32
kern_ioctl+0x3fc (?,?,?,?) ra ffffffff80401654 sp c000000028247790 sz 112
user_ioctl+0x1d4 (?,?,?,?) ra ffffffff8010a46c sp c000000028247800 sz 224
freebsd64_ioctl+0xd4 (?,?,?,?) ra ffffffff806b513c sp c0000000282478e0 sz 16
trap+0x1424 (?,?,?,?) ra ffffffff8069ca64 sp c0000000282478f0 sz 272
MipsUserGenException+0x1c0 (?,?,?,0) ra 0 sp c000000028247a00 sz 0
panic: vm_fault: fault on nofault entry, addr: 0xc000000028248000
cpuid = 1
```

brooksdavis, 2020-04-21:
Is this reproducible with a fresh kernel?

nwf, 2020-04-21:
I was not able to reproduce it even at the time; it is/was a very rare race or maybe just a cosmic ray.  I'll close the issue until it rears its head again.

closed 2020-04-21 by nwf

bsdjhb, 2020-04-21:
I suspect this might be an upstream issue, but one that is being looked at (e.g. D24214 and D24215) upstream.