CTSRD-CHERI/cheribsd #417: vm_page_assert_xbusied panic running libthr tests

arichardson, 2020-04-02:
To reproduce: `cheribuild.py run --test "--test-extra-args=--kyua-tests-files /usr/tests/lib/libthr/Kyuafile"` with current master.

```
root@qemu-test:~ # kyua test --results-file=/tmp/results.db -k /usr/tests/lib/libthr/Kyuafile
atexit_test:atexit  ->  passed  [0.456s]
barrier_test:barrier  ->  passed  [10.573s]
cancel_test:register_while_disabled  ->  passed  [0.388s]
cond_test:bogus_timedwaits  ->  passed  [0.079s]
cond_test:broadcast  ->  passed  [3.954s]
cond_test:cond_timedwait_race  ->  panic: vm_page_assert_xbusied: page 0x98000000fcf18c90 not exclusive busy @ /local/scratch/alr48/cheri/cheribsd/sys/vm/vm_glue.c:533
time = 1585781630
KDB: enter: panic
[ thread pid 640 tid 100042 ]
Stopped at      kdb_enter+0x94: break   0
db> bt
Tracing pid 640 tid 100042 td 0xc0000000002fea80
kdb_enter+0x94 (?,?,?,?) ra ffffffff804a79ec sp c00000001a137250 sz 32
vpanic+0x1f4 (?,?,?,?) ra ffffffff804a7670 sp c00000001a137270 sz 64
panic+0x60 (?,?,?,?) ra ffffffff80779308 sp c00000001a1372b0 sz 96
vm_thread_dispose+0x248 (?,?,?,?) ra ffffffff80779854 sp c00000001a137310 sz 96
kick_proc0+0x384 (?,?,?,?) ra ffffffff8076ff0c sp c00000001a137370 sz 48
sysctl_handle_uma_zone_cur+0x45c4 (?,?,?,?) ra ffffffff8076f804 sp c00000001a1373a0 sz 112
sysctl_handle_uma_zone_cur+0x3ebc (?,?,?,?) ra ffffffff807717c8 sp c00000001a137410 sz 48
sysctl_handle_uma_zone_cur+0x5e80 (?,?,?,?) ra ffffffff807684b4 sp c00000001a137440 sz 80
uma_zalloc_domain+0x414 (?,?,?,?) ra ffffffff80767b98 sp c00000001a137490 sz 64
uma_zfree_arg+0x390 (?,?,?,?) ra ffffffff8077911c sp c00000001a1374d0 sz 64
vm_thread_dispose+0x5c (?,?,?,?) ra ffffffff804c6da4 sp c00000001a137510 sz 16
thread_reap+0x15c (?,?,?,?) ra ffffffff804c7a90 sp c00000001a137520 sz 48
thread_wait+0x110 (?,?,?,?) ra ffffffff8044a178 sp c00000001a137550 sz 32
proc_reap+0x6a0 (?,?,?,?) ra ffffffff8044a8b8 sp c00000001a137570 sz 64
proc_reap+0xde0 (?,?,?,?) ra ffffffff80449568 sp c00000001a1375b0 sz 112
kern_wait6+0x2a8 (?,?,?,?) ra ffffffff80448eec sp c00000001a137620 sz 192
kern_wait4+0x8c (?,?,?,?) ra ffffffff801648f8 sp c00000001a1376e0 sz 512
freebsd64_wait4+0x130 (?,?,?,?) ra ffffffff807e63ec sp c00000001a1378e0 sz 16
trap+0xfe4 (?,?,?,?) ra ffffffff807d095c sp c00000001a1378f0 sz 272
```

brooksdavis, 2020-04-06:
Definitely something local.  Did a run on base freebsd mips:
```
cond_test:cond_timedwait_race  ->  passed  [14.322s]
```

arichardson, 2020-05-06:
Running `kyua test` on amd64 CheriBSD (in qemu-system-x86_64 with KVM acceleration) I get the same kind of backtrace:

```
lib/googletest/gmock/gmock_stress_test:main  ->  panic: vm_page_assert_xbusied: page 0xfffffe0002b26f40 not exclusive busy @ /local/scratch/alr48/cheri/cheribsd/sys/vm/vm_glue.c:533
cpuid = 0
time = 1588768082
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xfffffe0024ba8670
vpanic() at vpanic+0x182/frame 0xfffffe0024ba86c0
panic() at panic+0x5b/frame 0xfffffe0024ba8720
vm_thread_stack_dispose() at vm_thread_stack_dispose+0x11c/frame 0xfffffe0024ba8770
kstack_release() at kstack_release+0x46/frame 0xfffffe0024ba87a0
bucket_drain() at bucket_drain+0x20f/frame 0xfffffe0024ba87f0
bucket_free() at bucket_free+0x25/frame 0xfffffe0024ba8820
cache_free() at cache_free+0x36d/frame 0xfffffe0024ba8870
uma_zfree_arg() at uma_zfree_arg+0x297/frame 0xfffffe0024ba88c0
thread_reap() at thread_reap+0x102/frame 0xfffffe0024ba88e0
proc_reap() at proc_reap+0x659/frame 0xfffffe0024ba8920
proc_to_reap() at proc_to_reap+0x3ce/frame 0xfffffe0024ba8970
kern_wait6() at kern_wait6+0x1bb/frame 0xfffffe0024ba8a10
kern_wait4() at kern_wait4+0x76/frame 0xfffffe0024ba8c10
amd64_syscall() at amd64_syscall+0x140/frame 0xfffffe0024ba8d30
fast_syscall_common() at fast_syscall_common+0x101/frame 0xfffffe0024ba8d30
--- syscall (7, FreeBSD ELF64, sys_wait4), rip = 0x8015893ea, rsp = 0x7fffffffd4d8, rbp = 0x7fffffffd510 ---
KDB: enter: panic
[ thread pid 6932 tid 100070 ]
Stopped at      kdb_enter+0x37: movq    $0,0x10e9366(%rip)
db>
```

bsdjhb, 2020-05-06:
This was a mismerge of an earlier kstack related change.  I expect the branch I'm working on to deal with the kstack object change will fix this.

bsdjhb, 2020-05-06:
I believe this is fixed.

PR #460 (2020-05-06): kstack allocation cleanups
After running into the buzzsaw of trying to rebase my earlier merge fix for the kstack issue, I decided to instead break out most of the cleanups into an earlier set of commits.  Once this is done, the merge is much simpler to resolve.  This doesn't include the actual merge as I plan to do that and push it to dev after this has landed on dev.

This should also fix #417
closed 2020-05-06 by bsdjhb