CTSRD-CHERI/cheribsd #427: PT_SETREGS should not panic kernel due to untagged PCC

jrtc27, 2020-04-13:
Currently, calling `PT_SETREGS` to change PC(C) on MIPS can panic the kernel with:

```
panic: created untagged pcc: _update_pcc_offset: result v:0 s:0 p:00000000 f:0 b:0000000000000000 l:ffffffffffffffff o:4041e910 t:-1
```

We should permit this and just let userspace fault. This was encountered when trying to `set $pc=0x4041e910` in GDB in order to fix up PC for a `bt`.

brooksdavis, 2020-04-23:
It looks like this was added in f9e44f334302574cdccb99e837d1a527dc188fa7.

arichardson, 2020-04-23:
I _think_ it's fine to convert this assertion to a warning message. I assumed the callers would always be setting valid values, but it appears that they can be called with arbitrary values from userspace. We clearly shouldn't be panicking the kernel but should rather let the userspace process die on return.

I'm not sure if there are any callsites where the assertion makes sense since I don't really understand this part of the kernel.

rwatson, 2020-04-23:
I would argue that, just as it's OK to set PC to an arbitrary value (e.g., NULL) with the expectation of a signal being delivered in conventional architectures, installing a PCC that will fall is also fine in CHERI-land. The application itself may have a signal handler in place intended to catch and handle that.

arichardson, 2020-04-23:
I think I originally added this helper when I migrated MIPS from using EPC  to using epcc. But I think now all uses are from userspace so we should probably remove the assertion.

closed 2020-04-23 by brooksdavis