CTSRD-CHERI/cheribsd #447: Audit imprecise cheri_setbounds() use:

brooksdavis, 2020-04-30:
- [ ] libc
 - [ ] libthr
 - [ ] rtld
 - [ ] kernel - execve
 - [ ] kernel - overall

arichardson, 2020-04-30:
One thing I noticed in RTLD is that PCC (and in the MIPS PLT ABI CGP) allow accesses past the end of the captable due to representability.

The captable needs to be covered by PCC, but it has to be part of the data segment since we need to dynamically initialize it.
The range [start of rodata/text -> end of captable] is unlikely to be precisely representable, but LLD makes this a lot worse by aligning all segments to multiples of 0x10000.
We should probably reduce that alignment to pack the segments closer together.
For some reason the .captable section also ends up near the end of the data segment, so we always cover .data...
I'll open some LLD issues and try to improve the situation shortly.

jrtc27, 2020-04-30:
It aligns to 0x10000 because that's what the MIPS ABI requires due to page sizes. RISC-V stays at 0x1000.