CTSRD-CHERI/cheribsd #452: Verify FreeBSD/CheriBSD RISC-V comparability with VirtIO in tinyemu

brooksdavis, 2020-05-01:
https://bellard.org/tinyemu/

rwatson, 2020-05-04:
Various bits of progress here:

- @jrtc27 has done some initial experimentation and fixed some issues with FreeBSD’s VirtIO drivers and MMIO: https://reviews.freebsd.org/D24681
- @jrtc27 has confirmed that block storage can work, but very slowly, in simulation.
@jrtc27 has observed that we do not have support for the “linux,dma” feature that allows FDT to force bounce buffering via specific DMA-accessible memory, which we likely required. It may make sense for @bukinr to implement this. While access to on-FGPA memory is being done via the debug unit, access is coherent and doesn’t need to bounce buffer, but once actual DMA is done, it will be required.
- @bukinr is going to look at bringing up FreeBSD/RISC-V on TinyEmu to address potential VirtIO issues.

jrtc27, 2020-05-04:
> * @jrtc27 has confirmed that block storage can work, but very slowly, in simulation.

As discussed a few minutes ago, this was on an F1 FPGA, not in simulation (even if I/O speed feels like it must be in a simulation).

brooksdavis, 2020-05-13:
Given we have something working, I'm going to close this. IIRC we still have a hack in tinyemu for different virtio versions.  We should make tickets to clean that up in the appropriate place.

jrtc27, 2020-05-13:
I believe the only hack in tinyemu, at least for Connectal, is that the `num_buffers` field at the end of the virtio network request header was conditionally present for legacy devices based on the negotiation of one of the feature bits, but is now unconditionally present for non-legacy devices, and tinyemu has not negotiated the feature bit so FreeBSD believes the field isn't there. It should learn that version 2 also implies the existence of the field and that it should be set to 1 even if the feature bit isn't present (though tinyemu will ignore its value).

rwatson, 2020-05-21:
@bukinr: Could you take a look at fixing the FreeBSD VirtIO front end to do num_buffers negotiation properly? @jrtc27 and I suspect it involves slightly annoying passing of flag information between the MMIO layer and the VirtIO network parts, but is otherwise minor and boring.

jrtc27, 2020-05-21:
Specifically it's the VIRTIO_F_VERSION_1 feature flag, and `vtmmio_negotiate_features` means that both the driver and the virtqueue implementation need to

jrtc27, 2022-01-21:
FETT ran without any known VirtIO-related hitches in production

closed 2022-01-21 by jrtc27