CTSRD-CHERI/cheribsd #463: foreign cross build depends on removed MK_SYMVER

brooksdavis, 2020-05-08:
@bsdjhb removed MK_SYMVER upstream and tools/build/cross-build/libfreebsd/Makefile.inc depends on it.  I've reverted the change in an upcoming merge and in the freebsd-crossbuild branch, but we need to work out what we should replace it with.

brooksdavis, 2020-05-08:
Personally I'd like to have the ability to not include the libc compat code. I tried to sell the idea to upstream a few years ago and failed.

bsdjhb, 2020-05-08:
When someone tried it via the build options survey script, it didn't build due to trying to define versions for combat symbols for kevent_freebsd11.  Given how long ago those were added (several years), I assumed this meant WITHOUT_SYMVER hadn't succesfully built in many years and was hence unmaintained.  I'm surprised that cross build would need to turn it off?  You have to have a really old toolchain (pre-binutils 2.17) to not support symbol versioning and I don't think such an old toolchain ships on any current versions of OS X or Linux.

arichardson, 2020-05-08:
I guess those are just missing macros in the crossbuild compat cdefs.h. Will fix asap.

jrtc27, 2020-05-08:
Symbol versioning is far older than that, no? Binutils had code for it from the late 90s (I see a commit from 1997 to a tree that claimed it was ld 2.8.2 adding an unversioned fallback for default symbol lookups). Even 10, maybe 15, years ago any useful Linux system should have had it out of the box in its linker. macOS may be a different story if you're trying to link native libraries with its stubborn pursuance of Mach-O over ELF.

arichardson, 2020-05-08:
Seems to be building just fine on macos with 
```
#define	__sym_compat(sym,impl,verid) /* nothing */
#define	__sym_default(sym,impl,verid) /* nothing */
#define	__warn_references(sym,msg) /* nothing */
```

jrtc27, 2020-05-08:
Yeah, you don't actually need symbol versioning here given we're compiling everything from source, you just need to appease the source and build. FYI, macOS's "symbol versioning" approach seems to just be "add `__asm("foo$VERSION")` to force a different name", and expose this to the user's source code rather than making it a link-time decision. For example:
```c
int pselect(int, fd_set * restrict, fd_set * restrict,
    fd_set * restrict, const struct timespec * restrict,
    const sigset_t * restrict) __asm("_" "pselect" "$1050")
```

brooksdavis, 2020-05-27:
Also merged to freebsd-crossbuild in 507e6747ddb71ac767277a2fe6519ffb451700f3

PR #464 (2020-05-18): Fix crossbuild bootstrap after MK_SYMVER removal
Fixes #463
closed 2020-05-27 by brooksdavis