CTSRD-CHERI/cheribsd #468: Support Non-Legacy VirtIO Network Device

jrtc27, 2020-05-14:
5.1.6.1 Legacy Interface: Device Operation

> The legacy driver only presented num_buffers in the struct virtio_net_hdr when VIRTIO_NET_F_MRG_RXBUF was negotiated; without that feature the structure was 2 bytes shorter.

TinyEMU does not support VIRTIO_NET_F_MRG_RXBUF, so the feature is not negotiated and FreeBSD does not include the field in its header, resulting in all network packets being corrupted. It should learn that non-legacy (i.e. v1.0, encoded as the integer 2) network devices always include the field, even if the feature is not negotiated, and it should be set to 1 to be strictly spec-conforming (though TinyEMU will ignore it entirely).

jrtc27, 2020-06-04:
Fix submitted for review: https://reviews.freebsd.org/D25132 (and associated parent revision).

brooksdavis, 2020-06-15:
We've got one fix to loop back, but I think we're done for FETT purposes.

closed 2020-06-15 by brooksdavis