CTSRD-CHERI/cheribsd #497: riscv64-purecap file(1) crashes on riscv64-purecap /bin/csh

jrtc27, 2020-05-23:
```
root@qemu-cheri-jrtc4:~ # file /bin/sh
/bin/sh: ELF 64-bit LSB shared object, UCB RISC-V, version 1 (SYSV), dynamically linked, interpreter /libexec/ld-elf.so.1, for FreeBSD 13.0 (1300092), FreeBSD-style, with debug_info, not stripped
root@qemu-cheri-jrtc4:~ # file /bin/csh
/bin/csh: Abort (core dumped)
```

(Separately, we might wish to teach file(1) about our RISC-V ELF flags)

bsdjhb, 2020-05-26:
Appears to be an internal crash in malloc(), perhaps from the recent malloc hardening?

```
#0  thr_kill () at thr_kill.S:4
#1  0x00000000403351d8 in __raise (s=<optimized out>)
    at /usr/home/john/work/cheri/git/cheribsd/lib/libc/gen/raise.c:52
#2  0x00000000402a50f0 in abort ()
    at /usr/home/john/work/cheri/git/cheribsd/lib/libc/stdlib/abort.c:67
#3  0x0000000040290bc4 in unbound_ptr (tsdn=<optimized out>, 
    ptr=<optimized out>)
    at /usr/home/john/work/cheri/git/cheribsd/contrib/jemalloc/include/jemalloc/internal/jemalloc_internal_inlines_c.h:49
#4  idalloctm (tsdn=<optimized out>, ptr=<optimized out>, 
    tcache=<optimized out>, alloc_ctx=<optimized out>, 
    is_internal=<optimized out>, slow_path=<optimized out>)
    at /usr/home/john/work/cheri/git/cheribsd/contrib/jemalloc/include/jemalloc/internal/jemalloc_internal_inlines_c.h:175
#5  a0idalloc (ptr=<optimized out>, is_internal=<optimized out>)
    at jemalloc_jemalloc.c:285
#6  0x000000004029119e in __je_a0dalloc (ptr=<optimized out>)
    at jemalloc_jemalloc.c:295
#7  __je_arena_tdata_get_hard (tsd=<optimized out>, ind=<optimized out>)
    at jemalloc_jemalloc.c:541
#8  0x00000000402624a8 in arena_tdata_get (
    tsd=0x407da040 [rwRW,0x407da010-0x407dc2a0], ind=<optimized out>, 
    refresh_if_missing=true)
    at /usr/home/john/work/cheri/git/cheribsd/contrib/jemalloc/include/jemalloc/internal/jemalloc_internal_inlines_a.h:73
#9  decay_ticker_get (tsd=0x407da040 [rwRW,0x407da010-0x407dc2a0], 
    ind=<optimized out>)
    at /usr/home/john/work/cheri/git/cheribsd/contrib/jemalloc/include/jemalloc/internal/jemalloc_internal_inlines_a.h:104
#10 arena_decay_ticks (tsdn=0x407da040 [rwRW,0x407da010-0x407dc2a0], 
    arena=<optimized out>, nticks=1)
    at /usr/home/john/work/cheri/git/cheribsd/contrib/jemalloc/include/jemalloc/internal/arena_inlines_b.h:132
#11 arena_decay_tick (tsdn=0x407da040 [rwRW,0x407da010-0x407dc2a0], 
    arena=<optimized out>)
    at /usr/home/john/work/cheri/git/cheribsd/contrib/jemalloc/include/jemalloc/internal/arena_inlines_b.h:146
#12 __je_large_palloc (tsdn=0x407da040 [rwRW,0x407da010-0x407dc2a0], 
    arena=<optimized out>, usize=<optimized out>, alignment=<optimized out>, 
    zero=<optimized out>) at jemalloc_large.c:81
#13 0x0000000040294fbc in arena_malloc (tsdn=<optimized out>, 
    arena=<optimized out>, size=<optimized out>, ind=<optimized out>, 
    zero=<optimized out>, tcache=<optimized out>, slow_path=<optimized out>)
    at /usr/home/john/work/cheri/git/cheribsd/contrib/jemalloc/include/jemalloc/internal/arena_inlines_b.h:187
#14 iallocztm (tsdn=<optimized out>, size=<optimized out>, 
    ind=<optimized out>, zero=<optimized out>, tcache=<optimized out>, 
    is_internal=<optimized out>, arena=<optimized out>, 
    slow_path=<optimized out>)
    at /usr/home/john/work/cheri/git/cheribsd/contrib/jemalloc/include/jemalloc/internal/jemalloc_internal_inlines_c.h:110
#15 imalloc_no_sample (sopts=<optimized out>, dopts=<optimized out>, 
    tsd=<optimized out>, size=<optimized out>, usize=<optimized out>, 
    ind=<optimized out>) at jemalloc_jemalloc.c:2003
#16 imalloc_body (sopts=<optimized out>, dopts=<optimized out>, 
    tsd=<optimized out>) at jemalloc_jemalloc.c:2203
#17 imalloc (sopts=<optimized out>, dopts=<optimized out>)
    at jemalloc_jemalloc.c:2325
#18 __calloc (num=<optimized out>, size=<optimized out>)
    at jemalloc_jemalloc.c:2577
#19 0x000000004013eab2 in file_encoding (ms=<optimized out>, 
    b=<optimized out>, ubuf=<optimized out>, ulen=<optimized out>, 
    code=<optimized out>, code_mime=<optimized out>, type=<optimized out>)
    at /usr/home/john/work/cheri/git/cheribsd/contrib/file/src/encoding.c:91
#20 0x000000004013ba1c in file_buffer (ms=<optimized out>, fd=<optimized out>, 
    st=<optimized out>, inname=<optimized out>, buf=<optimized out>, 
    nb=<optimized out>)
    at /usr/home/john/work/cheri/git/cheribsd/contrib/file/src/funcs.c:234
#21 0x0000000040133cbc in file_or_fd (ms=<optimized out>, 
    inname=<optimized out>, fd=<optimized out>)
    at /usr/home/john/work/cheri/git/cheribsd/contrib/file/src/magic.c:514
#22 0x0000000000106b30 in process (ms=<optimized out>, inname=<optimized out>, 
    wid=<optimized out>)
    at /usr/home/john/work/cheri/git/cheribsd/contrib/file/src/file.c:542
#23 0x0000000000105e2a in main (argc=<optimized out>, argv=<optimized out>)
    at /usr/home/john/work/cheri/git/cheribsd/contrib/file/src/file.c:413
(gdb) frame 3
#3  0x0000000040290bc4 in unbound_ptr (tsdn=<optimized out>, 
    ptr=<optimized out>)
    at /usr/home/john/work/cheri/git/cheribsd/contrib/jemalloc/include/jemalloc/internal/jemalloc_internal_inlines_c.h:49
49                      malloc_write("<jemalloc>: refusing to unbound cap at "
```

Given that malloc_write supposedly ends up calling write(2) to stderr, this seems a bit odd.  I guess it might use syscall(2) directly on RISC-V and maybe that doesn't work?

bsdjhb, 2020-05-26:
Here's the call to write() that is supposed to log the message:

```
   785 file.cheri CALL  write(0x40262130,0x40175d91,0)
   785 file.cheri RET   write -1 errno 9 Bad file descriptor
```

The first pointer used as the syscall arg is the address of syscall():

```
(gdb) x/i 0x40262130
   0x40262130 <syscall>:        li      t0,0
```

I suspect the var-args handling for syscall is just broken.  How do var-args work for purecap on RISC-V, is it the same as on MIPS?

jrtc27, 2020-05-26:
Yes and no. All varargs are forced onto the stack even if registers are free (like CHERI-MIPS), but currently you probe the stack directly rather than a special register.

brooksdavis, 2020-05-29:
Merging #505 fixed this.

closed 2020-05-29 by brooksdavis