CTSRD-CHERI/cheribsd #498: RISC-V ping(8) gets stuck after first response

jrtc27, 2020-05-23:
Seen with both freebsd64 and purecap userland:

```
root@qemu-cheri-jrtc4:~ # ping 10.0.2.2                                                        
PING 10.0.2.2 (10.0.2.2): 56 data bytes                               
64 bytes from 10.0.2.2: icmp_seq=0 ttl=255 time=6.515 ms             
ping: recvmsg: Invalid argument
ping: recvmsg: Invalid argument
ping: recvmsg: Invalid argument
...
```

(repeats flooding the terminal until Ctrl-C'ed)

trasz, 2020-05-31:
So the fix is basically:

```
--- a/sys/kern/uipc_syscalls.c
+++ b/sys/kern/uipc_syscalls.c
@@ -1208,6 +1208,7 @@ sys_recvmsg(struct thread *td, struct recvmsg_args *uap)
 {
        struct msghdr msg;
        struct iovec *iov;
+       struct iovec * __capability uiov;
        int error;
 
        error = copyincap(uap->msg, &msg, sizeof(msg));
@@ -1216,6 +1217,7 @@ sys_recvmsg(struct thread *td, struct recvmsg_args *uap)
        error = copyiniov(msg.msg_iov, msg.msg_iovlen, &iov, EMSGSIZE);
        if (error != 0)
                return (error);
+       uiov = msg.msg_iov;
        msg.msg_iov = (__cheri_tocap struct iovec * __capability)iov;
        msg.msg_flags = uap->flags;
 #ifdef COMPAT_OLDSOCK
@@ -1223,8 +1225,10 @@ sys_recvmsg(struct thread *td, struct recvmsg_args *uap)
                msg.msg_flags &= ~MSG_COMPAT;
 #endif
        error = recvit(td, uap->s, &msg, NULL);
-       if (error == 0)
+       if (error == 0) {
+               msg.msg_iov = uiov;
                error = copyoutcap(&msg, uap->msg, sizeof(msg));
+       }
        free(iov, M_IOV);
        return (error);
 }
```

What's left to do is to figure out why it got removed; should it be preserved in some other way?

jrtc27, 2020-05-31:
I assume just an oversight given none of the other syscalls need to save `msg_iov`, so this was erroneously made to look like them without looking at the upstream code. I'll file a PR to do this properly (we now have some needless diffs from upstream).

closed 2020-06-04 by jrtc27