CTSRD-CHERI/cheribsd #525: ldd doesn't work on RISC-V CheriABI binaries

brooksdavis, 2020-06-04:
```
root@qemu-cheri-bed22:~ # ldd /bin/sh
ldd: /bin/sh: not a FreeBSD ELF shared object
```

jrtc27, 2020-06-04:
I think we determined this was because LLD isn't setting EI_OSABI and so the check for ELFOSABI_FREEBSD fails.

jrtc27, 2020-06-04:
No CHERI involved:

```
root@qemu-cheri-jrtc4:~ # uname -a
FreeBSD qemu-cheri-jrtc4 13.0-CURRENT FreeBSD 13.0-CURRENT #0 770515395e89-c326253(dev): Fri Jun  5 00:50:51 BST 2020     jrtc4@zeno.sec.cl.cam.ac.uk:/home/jrtc4/cheri/build/cheribsd-riscv64-build/home/jrtc4/cheri/cheribsd/riscv.riscv64/sys/QEMU  riscv
root@qemu-cheri-jrtc4:~ # ldd /bin/sh
ldd: /bin/sh: not a FreeBSD ELF shared object
```

jrtc27, 2020-06-05:
The same is going to be true even for GCC:

```
jrtc4@zeno:~/tmp/osabi$ cat start.c
void _start(void) { }
jrtc4@zeno:~/tmp/osabi$ riscv64-unknown-freebsd12.0-gcc -B/usr/local/riscv64-unknown-freebsd12.0/bin -o start start.c -nostdlib
jrtc4@zeno:~/tmp/osabi$ riscv64-unknown-freebsd12.0-readelf -h start
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           RISC-V
  Version:                           0x1
  Entry point address:               0x10078
  Start of program headers:          64 (bytes into file)
  Start of section headers:          608 (bytes into file)
  Flags:                             0x5, RVC, double-float ABI
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         1
  Size of section headers:           64 (bytes)
  Number of section headers:         6
  Section header string table index: 5
```

Neither GCC nor binutils have knowledge of elf64lriscv_fbsd; binutils just falls back on the generic riscv64*-*-* case, and GCC just passes elf64lriscv (I haven't looked at whether it specifically requests it or it's a default).

Clang's FreeBSD driver thus explicitly requests elf64lriscv. It could request elf64lriscv_fbsd, and lld would automatically do the right thing (it just looks for and strips the suffix, regardless of the emulation), but then that would break Clang out of the box if it's defaulting to BFD.

This is thus a universal FreeBSD on RISC-V problem that applies to every single toolchain out there. I find it surprising that nobody has tried running ldd in all these years...

jrtc27, 2020-06-05:
Ah, right, this is only checked for shared objects, which we're hitting because we compile everything as PIE by default.

jrtc27, 2020-06-05:
I think AArch64 will encounter this too, as Clang doesn't pass a specific emulation for it. Moreover the AArch64 ELF ABI mandates that EI_OSABI be ELFOSABI_NONE, with the only exception permitted being if you have OS-specific meanings to flags in your file.

I think ldd should be less draconian about a field that's almost always a complete waste of time/space. ELFOSABI_NONE should be allowed everywhere, and the `#ifdef __ARM_EABI__` should move to only cover the ARM machine and ABI checks.

brooksdavis, 2020-06-05:
@emaste This probably impacts plans to make WITH_PIE the default in FreeBSD

jrtc27, 2020-06-05:
Yeah, ldd on shared libraries (admittedly an uncommon thing to do) will be broken upstream, and WITH_PIE just makes the situation worse and more apparent.

bsdjhb, 2020-06-08:
GDB checks for either the OSABI or an ABI tag note.  ldd should probably add the check for the ABI tag.  I ran into this for aarch64 kernels as well for kgdb.

bsdjhb, 2020-06-08:
I'm working on this one.  My candidate patch uses gelf(3) from libelf to reduce some code duplication and should support PIE binaries on all architectures.

bsdjhb, 2021-01-27:
https://reviews.freebsd.org/D28342 is the upstream review.

closed 2021-04-20 by bsdjhb