CTSRD-CHERI/cheribsd #526: ssh rejects valid ecdsa host keys

brooksdavis, 2020-06-05:
The base system ssh rejects ecdsa host keys (the FETT version and the amd64 build system think they are fine):
```
root@cheri-fett:~ # /usr/bin/ssh-keygen -l -f /etc/ssh/ssh_host_ecdsa_key
/etc/ssh/ssh_host_ecdsa_key is not a key file.
root@cheri-fett:~ # /fett/bin/ssh-keygen -l -f /etc/ssh/ssh_host_ecdsa_key
256 SHA256:4j4O2s5I2COg/z8cCaFBghHOBkyFL2CK9P1bda6WY6k bed22@baume.cl.cam.ac.uk (ECDSA)
```
It seems likely this is a miscompilation, but I've made no effort to diagnose the issue further.

jrtc27, 2020-06-05:
Have you tried a plain old x86 ssh-keygen (either FreeBSD's or a random Linux box's)?

brooksdavis, 2020-06-05:
The x86 ssh-keygen produces the same result at the fett one.

brooksdavis, 2020-06-05:
These keys are generated on the x86 host and stuffed into the image.

arichardson, 2020-06-07:
I'm seeing this too for riscv64-purecap.
```
Error loading host key "/etc/ssh/ssh_host_ecdsa_key": invalid format
Error loading host key "/etc/ssh/ssh_host_ecdsa_key": invalid format
Could not load host key: /etc/ssh/ssh_host_ecdsa_key
```
On MIPS and x86 this key seems to be fine, so I agree that a miscompilation of RISC-V ssh is the likely cause.

arichardson, 2020-06-18:
I just compiled libcrypto+ssh-keygen at -O0 and it works as expected. Sounds like a miscompilation to me.

jrtc27, 2020-06-18:
What about riscv64-nocheri/hybrid?

arichardson, 2020-06-18:
Hybrid works, I didn't test nocheri.

jrtc27, 2020-06-18:
If hybrid works then there's a 99.999999% chance nocheri does too. So definitely a riscv64-purecap issue. Was your MIPS test mips-purecap or mips-nocheri/hybrid?

arichardson, 2020-06-18:
Both work for MIPS.

arichardson, 2020-06-21:
I get the error if I compiler libcrypto at -O2, but both -O0 and -O1 work as expected.

arichardson, 2020-06-22:
Compiling https://github.com/CTSRD-CHERI/cheribsd/blob/dev/crypto/openssl/crypto/bn/bn_nist.c at -O0 fixes the issue.

jrtc27, 2020-06-22:
Should be able to use the bugpoint miscompilation debugger on that, just need to extract the right flags (and link command) to be able to slot it into the rest of the .o's?

arichardson, 2020-06-22:
Yes that should work. I'll give it a go now.

arichardson, 2020-06-22:
Finally got something out of bugpoint:

Good command: `opt -mtriple=riscv64-unknown-freebsd -target-abi l64pc128d -mattr=+xcheri,+cap-mode -tbaa -gvn -stats -debug -gvn-max-num-deps=0 bugpoint-funcs-tooptimize.ll -S -o bugpoint-funcs-tooptimize.good.ll`
Bad command: `opt -mtriple=riscv64-unknown-freebsd -target-abi l64pc128d -mattr=+xcheri,+cap-mode -tbaa -gvn -stats -debug -gvn-max-num-deps=1 bugpoint-funcs-tooptimize.ll -S -o bugpoint-funcs-tooptimize.bad.ll`

```
--- bugpoint-funcs-tooptimize.good.ll	2020-06-22 21:51:21.000000000 +0100
+++ bugpoint-funcs-tooptimize.bad.ll	2020-06-22 21:51:22.000000000 +0100
@@ -197,30 +197,15 @@
   %conv28 = zext i32 %tmp10 to i64
   %add = add nuw nsw i64 %conv26, %conv28
   %arrayidx29 = getelementptr inbounds i32, i32 addrspace(200)* %arraydecay25, i64 1
-  %tmp11 = load i32, i32 addrspace(200)* %arrayidx29, align 4, !tbaa !13
-  %conv30 = zext i32 %tmp11 to i64
-  %add31 = add nuw nsw i64 %add, %conv30
   %arrayidx32 = getelementptr inbounds i32, i32 addrspace(200)* %arraydecay25, i64 3
-  %tmp12 = load i32, i32 addrspace(200)* %arrayidx32, align 4, !tbaa !13
-  %conv33 = zext i32 %tmp12 to i64
-  %sub34 = sub nsw i64 %add31, %conv33
   %arrayidx35 = getelementptr inbounds %2, %2 addrspace(200)* %buf, i64 0, i32 0, i64 2
   %tmp13 = bitcast i64 addrspace(200)* %arrayidx35 to i32 addrspace(200)*
-  %tmp14 = load i32, i32 addrspace(200)* %tmp13, align 8, !tbaa !13
-  %conv36 = zext i32 %tmp14 to i64
-  %sub37 = sub nsw i64 %sub34, %conv36
   %arrayidx38 = getelementptr inbounds i32, i32 addrspace(200)* %arraydecay25, i64 5
-  %tmp15 = load i32, i32 addrspace(200)* %arrayidx38, align 4, !tbaa !13
-  %conv39 = zext i32 %tmp15 to i64
-  %sub40 = sub nsw i64 %sub37, %conv39
   %arrayidx41 = getelementptr inbounds %2, %2 addrspace(200)* %buf, i64 0, i32 0, i64 3
   %tmp16 = bitcast i64 addrspace(200)* %arrayidx41 to i32 addrspace(200)*
-  %tmp17 = load i32, i32 addrspace(200)* %tmp16, align 8, !tbaa !13
-  %conv42 = zext i32 %tmp17 to i64
-  %sub43 = sub nsw i64 %sub40, %conv42
-  %conv44 = trunc i64 %sub43 to i32
+  %conv44 = trunc i64 %add to i32
   store i32 %conv44, i32 addrspace(200)* %tmp8, align 4, !tbaa !13
-  %shr = ashr i64 %sub43, 32
+  %shr = ashr i64 %add, 32
   %arrayidx46 = getelementptr inbounds i32, i32 addrspace(200)* %tmp8, i64 1
   %tmp18 = load i32, i32 addrspace(200)* %arrayidx46, align 4, !tbaa !13
   %conv47 = zext i32 %tmp18 to i64
```

Input IR:
```
; ModuleID = '<stdin>'
source_filename = "reproducer.preprocessed.c"
target datalayout = "e-m:e-pf200:128:128:128:64-p:64:64-i64:64-i128:128-n64-S128-A200-P200-G200"
target triple = "riscv64-unknown-freebsd"

%0 = type { i64 addrspace(200)*, i32, i32, i32, i32 }
%1 = type opaque
%2 = type { [4 x i64] }
%3 = type { i64 (i64 addrspace(200)*, i64 addrspace(200)*, i64 addrspace(200)*, i32) addrspace(200)* }

@_bignum_nist_p_192 = external dso_local addrspace(200) constant %0, align 16
@_bignum_nist_p_224 = external dso_local addrspace(200) constant %0, align 16
@_bignum_nist_p_256 = external dso_local addrspace(200) constant %0, align 16
@_bignum_nist_p_384 = external dso_local addrspace(200) constant %0, align 16
@_bignum_nist_p_521 = external dso_local addrspace(200) constant %0, align 16
@BN_nist_mod_192._bignum_nist_p_192_sqr = external dso_local addrspace(200) constant %0, align 16
@_nist_p_192_sqr = external dso_local addrspace(200) constant [6 x i64], align 8
@_nist_p_192 = external dso_local addrspace(200) constant [3 x [3 x i64]], align 8
@BN_nist_mod_224._bignum_nist_p_224_sqr = external dso_local addrspace(200) constant %0, align 16
@_nist_p_224_sqr = external dso_local addrspace(200) constant [7 x i64], align 8
@_nist_p_224 = external dso_local addrspace(200) constant [2 x [4 x i64]], align 8
@BN_nist_mod_256._bignum_nist_p_256_sqr = external dso_local addrspace(200) constant %0, align 16
@_nist_p_256_sqr = external dso_local addrspace(200) constant [8 x i64], align 8
@_nist_p_256 = external dso_local addrspace(200) constant [5 x [4 x i64]], align 8
@BN_nist_mod_384._bignum_nist_p_384_sqr = external dso_local addrspace(200) constant %0, align 16
[... 536 lines cut ...]
attributes #0 = { nounwind uwtable "correctly-rounded-divide-sqrt-fp-math"="false" "disable-tail-calls"="false" "frame-pointer"="none" "less-precise-fpmad"="false" "min-legal-vector-width"="0" "no-infs-fp-math"="false" "no-jump-tables"="false" "no-nans-fp-math"="false" "no-signed-zeros-fp-math"="false" "no-trapping-math"="false" "stack-protector-buffer-size"="8" "target-features"="+a,+c,+cap-mode,+d,+f,+m,+xcheri,-relax,-save-restore,+xcheri,+cap-mode" "unsafe-fp-math"="false" "use-soft-float"="false" }
attributes #1 = { argmemonly nounwind willreturn }
attributes #2 = { "correctly-rounded-divide-sqrt-fp-math"="false" "disable-tail-calls"="false" "frame-pointer"="none" "less-precise-fpmad"="false" "no-infs-fp-math"="false" "no-nans-fp-math"="false" "no-signed-zeros-fp-math"="false" "no-trapping-math"="false" "stack-protector-buffer-size"="8" "target-features"="+a,+c,+cap-mode,+d,+f,+m,+xcheri,-relax,-save-restore,+xcheri,+cap-mode" "unsafe-fp-math"="false" "use-soft-float"="false" }
attributes #3 = { nounwind readnone willreturn }
attributes #4 = { nounwind }

!llvm.module.flags = !{!0, !1, !2}
!llvm.ident = !{!3}

!0 = !{i32 1, !"wchar_size", i32 4}
!1 = !{i32 1, !"target-abi", !"l64pc128d"}
!2 = !{i32 1, !"SmallDataLimit", i32 8}
!3 = !{!"clang version 11.0.0 "}
!4 = !{!5, !9, i64 16}
!5 = !{!"bignum_st", !6, i64 0, !9, i64 16, !9, i64 20, !9, i64 24, !9, i64 28}
!6 = !{!"any pointer", !7, i64 0}
!7 = !{!"omnipotent char", !8, i64 0}
!8 = !{!"Simple C/C++ TBAA"}
!9 = !{!"int", !7, i64 0}
!10 = !{!5, !6, i64 0}
!11 = !{!12, !12, i64 0}
!12 = !{!"long", !7, i64 0}
!13 = !{!9, !9, i64 0}
!14 = !{!7, !7, i64 0}
```

jrtc27, 2020-06-22:
FYI you should drop the d from the ABI or add +f,+d to mattr otherwise weird things could happen... though presumably you get away with it since there are no floats in this code.

arichardson, 2020-06-23:
Problem goes away when I compile with -fno-strict-aliasing

arichardson, 2020-06-23:
MIPS is not affected because the four lines that are (incorrectly) optimized out are guarded by a define that is only set for little endian.
Debug info from -fverbose-asm shows that lines bn_nist.c:725 ->  bn_nist:729:13 are missing when running with `opt -tbaa -gvn -gvn-max-num-deps=1`  but remain if we use `-gvn-max-num-deps=0`.

```
        acc += bp[8 - 8];
        acc += bp[9 - 8];
        acc -= bp[11 - 8];
        acc -= bp[12 - 8];
``` 

Well that was a waste of time 😭 OpenSSL 😡

closed 2020-07-29 by brooksdavis