CTSRD-CHERI/cheribsd #538: CheriBSD Vulnerability List

qwattash, 2020-06-12:
This is a place to gather interesting FreeBSD advisories that are relevant to CHERI.
The main purpose of this list is to support the effort of forward-porting interesting bugs and demonstrate that CHERI would mitigate or have some effect on them.

**Kernel Vulnerabilities**:
- CVE-2020-7452 SA-20:07 resolved by 3a5ef1b03378fae9c5c93f86a2368b9d35009990
  epair(4) interface allows to inject an user-controlled kernel pointer.
  The clone is created via SIOCIFCREATE2, which takes the param argument of
  `int epair_clone_create(struct if_clone *ifc, char *name, size_t len, void * __capability params)` which is abused to carry a `struct epair_softc *` and is probably assumed that will be `NULL` from the ioctl. 

**Userland Vulnerabilities**:

rwatson, 2020-06-15:
On CVE-2020-7452 SA-20:07, it's not clear to me that we have a lot to say about mitigation unless we are stripping the tag during copyin/out of the ioctl argument structure. Which I guess ideally we would be doing if it wasn't specified to include pointers explicitly by the device driver... Although perhaps that is not information known at the right moment, or requires too much annotation of drivers?

brooksdavis, 2020-06-15:
ioctl always copies capabilities in and out because it can't tell from the command encoding if it needs to. FreeBSD's centralization of the copyin/copyout bites us badly enough here that I think we may want to move away from it.

closed 2020-09-30 by brooksdavis