CTSRD-CHERI/cheribsd #554: mount_smbfs extremely unreliable on RISC-V

arichardson, 2020-06-17:
This is making testing more awkward than it should be.
SMBFS seems to regularly time out/get read errors when copying from a mounted directory.
I don't see these issues with MIPS, so it may be related to virtio-net.

arichardson, 2020-06-17:
It seems like running scp also sometimes suffers from issues. Around 1 in 10 times it fails.

arichardson, 2020-06-18:
I verified that it works as expected on an x86 image with virtio-net, so some bug related to RISCV interrupts sounds plausible.

jrtc27, 2020-06-18:
We believe CTSRD-CHERI/qemu@da1c25f4065814a430295d770708c33feb63eec9 would cause this, so hopefully that's what we were hitting. Please see if it still occurs?

arichardson, 2020-06-20:
Things seem a lot more reliable, but I am still seeing some freezes (but much less frequently).
I also just got this error with latest QEMU:
```
[PEXPECT\PROMPT]>mv -f /tmp/cheritest-mt.xml /test-results/cheritest-mt.xml ;if test $? -eq 0; then echo '__COMMAND' 'SUCCESSFUL__'; else echo '__COMMAND' 'FAILED__'; fi
mv: smb_iod_recvall: drop resp with mid 179
fastcopy: write() failed: /test-results/cheritest-mt.xml: Operation timed out
mv: /test-results/cheritest-mt.xml: remove: Operation not permitted
__COMMAND FAILED__
```

arichardson, 2020-06-21:
At least for me I still sometimes get hangs using tab completion on smbfs mounted dirs.

Currently the following is reproducible for me:
```
$ cheribuild.py run-riscv64-purecap
.....

# qemu-mount-rootfs.sh
# /nfsroot/usr/bin/ssh-key<TAB>
```