CTSRD-CHERI/cheribsd #735: Check permissions in vm_fault_quick_hold_pages

brooksdavis, 2020-10-15:


jrtc27, 2020-10-15:
Or if they're capabilities just and the permissions for the pages?

brooksdavis, 2020-10-16:
I think we don't want to do that because we'd potentially create a panic vector.

jrtc27, 2020-10-16:
Yeah, I came to that conclusion myself. I think we probably want to do _both_ for safety, i.e. have the checks be what should catch everything, and the permissions in case any API consumer tries to do more than it should (or the checks have a bug and are insufficient)? Same goes for bounds.

jrtc27, 2020-10-16:
Although one could restrict the safety net to just being for a pure-capability kernel, and treat it as now a kernel memory safety problem once you have the kernel mappings, given it'll likely be painful to make those all capabilities in a hybrid kernel.

brooksdavis, 2020-10-16:
I agree that doing both might be sensible in purecap.

@bsdjhb also pointed out that the page pointers created are physical addresses so we probably need physical address capabilities to restrict them properly.

jrtc27, 2020-10-16:
Hm it's worse than that, you get pointers to the `vm_page_t`'s. I naively assumed this was doing the lookup and pmap all at once...

bsdjhb, 2020-10-19:
In almost all cases, the reason callers use this function is to avoid mapping the pages into the kernel's address space at all.  Generally you are just going to pass the pages down to a driver that will map them for DMA, so it just pulls the physical address directly out of the `vm_page_t` via `VM_PAGE_TO_PHYS`.  For the few places that need a mapping, they are using the architecture-specific direct mapping of RAM and access one page at a time, or they use the sf_buf(9) API to temporarily map pages (which uses the direct mapping when it exists)

commit 094ad009 (2020-10-21): vm_fault_quick_hold_pages: check cap permissions

Add a vm_cap_allows_prot() to check if a capability has permissions
matching a given vm_prot_t and use it in vm_fault_quick_hold_pages.

Also use vm_cap_allows_prot in useracc and as a result drop the
requirement that the capability be global.  There is reason to belive
that most capabilities will be local in a co-process world.

Fixes #735
commit 2f753e62 (2020-11-04): vm_fault_quick_hold_pages: check cap permissions

Add a vm_cap_allows_prot() to check if a capability has permissions
matching a given vm_prot_t and use it in vm_fault_quick_hold_pages.

Also use vm_cap_allows_prot in useracc and as a result drop the
requirement that the capability be global.  There is reason to belive
that most capabilities will be local in a co-process world.

Fixes #735
PR #742 (2020-11-26): vm_fault_quick_hold_pages: check cap permissions
Add a vm_cap_allows_prot() to check if a capability has permissions
matching a given vm_prot_t.

Also use vm_cap_allows_prot in useracc and as a result drop the
requirement that the capability be global.  There is reason to belive
that most capabilities will be local in a co-process world.

Fixes #735

---
This change is [[image]](https://reviewable.io/reviews/ctsrd-cheri/cheribsd/742)
commit 48bca9e6 (2020-11-26): vm_fault_quick_hold_pages: check cap permissions

Add a vm_cap_allows_prot() to check if a capability has permissions
matching a given vm_prot_t and use it in vm_fault_quick_hold_pages.

Also use vm_cap_allows_prot in useracc and as a result drop the
requirement that the capability be global.  There is reason to belive
that most capabilities will be local in a co-process world.

Fixes #735
closed 2020-11-30 by brooksdavis