CTSRD-CHERI/cheribsd #786: Mismatched brackets in cheric.h.

jacobbramley, 2020-11-11:
I noticed `cheri_codeptr` trying to use it. Vim's syntax highlighting found `cheri_setbounds_changetype`.

https://github.com/CTSRD-CHERI/cheribsd/blob/ecbc5cc44c5e845074836fe92c2eb8fc100862d2/sys/cheri/cheric.h#L79
https://github.com/CTSRD-CHERI/cheribsd/blob/ecbc5cc44c5e845074836fe92c2eb8fc100862d2/sys/cheri/cheric.h#L122

arichardson, 2020-11-11:
We probably want to update cheric.h to use the newer cheriintrin.h macros, but those should definitely be fixed.

jrtc27, 2020-11-11:
The first one (and its friend below) should Justin be removed now we have the polymorphic builtins, though not sure we used those weird versions even before.

jrtc27, 2022-01-21:
Fixed in #797

PR #797 (2020-11-14): Tidy cheric.h
fixes #786
closed 2022-01-21 by jrtc27