CTSRD-CHERI/cheribsd #962: backtrace CHERI tag violation

nwf, 2021-03-18:
I am to the point that snmalloc compiles on CHERI again but am doing something wrong in startup.  On its way out, it tries to generate a stack trace by calling backtrace_symbols_fd and that, in turn, dies a CHERI death:

```
Trapframe Register Dump:
$0: 0                  at: 0x1                v0: 0x41c8a100         v1: 0xfffffffffffffffc
a0: 0x41c8a100         a1: 0x100              a2: 0x3                a3: 0
a4: 0                  a5: 0x2                a6: 0x390              a7: 0x390
t0: 0x28               t1: 0x7000000d         t2: 0x22               t3: 0x22
s0: 0x5                s1: 0x5                s2: 0x2                s3: 0x25
s4: 0                  s5: 0x1                s6: 0                  s7: 0
t8: 0                  t9: 0                  k0: 0                  k1: 0
gp: 0                  sp: 0                  s8: 0                  ra: 0
status: 0x408084b3 mullo: 0xf0; mulhi: 0; badvaddr: 0x40b7b560
cause: 0x48; pc: 0x40b93a74
BadInstr: 0xd8c10008 clc        c6,zero,128(c1)
CHERI cause: ExcCode: 0x02 RegNum: $c01 (tag violation)
$ddc: 0x0000000000000000
$c01: 0x0000000041c8a100 [rwRW,0x0000000041c8a200-0x0000000041c8a2f0] (invalid)
$c02: 0x0000000041c8a3a0 [rwRW,0x0000000041c8a300-0x0000000041c8a3f0]
$c03: 0x0000000041c82080 [rwRW,0x0000000041c82000-0x0000000041c820d0]
$c04: 0x0000000041c8a300 [rwRW,0x0000000041c8a300-0x0000000041c8a3f0]
$c05: 0x0000000041c8a101 [rwRW,0x0000000041c8a200-0x0000000041c8a2f0] (invalid)
$c06: 0x0000000000000001 [rwRW,0xffffffffffffe200-0xffffffffffffe2f0] (invalid)
$c07: 0x0000000041c8a100 [rwRW,0x0000000041c8a100-0x0000000041c8a1f0]
$c08: 0x0000007ffffd9750 [rwRW,0x0000007ffffd9750-0x0000007ffffd9760]
$c09: 0x0000007ffffd9740 [rwRW,0x0000007ffffd9740-0x0000007ffffd9750]
$c10: 0x0000007ffffcdb80 [rwRW,0x0000007ffffcdb80-0x0000007ffffcdb90]
$c11: 0x0000007ffffda440 [rwRW,0x0000007ffbfe0000-0x0000007ffffe0000]
$c12: 0x0000000040b93a30 [rxR,0x0000000040b78000-0x0000000040bc9a00] (sentry)
$c13: 0x0000000000000000
$c14: 0x0000007ffffcdb50 [rwRW,0x0000007ffffcdb50-0x0000007ffffcdb60]
$c15: 0x0000007ffffcdb40 [rwRW,0x0000007ffffcdb40-0x0000007ffffcdb50]
$c16: 0x0000007ffffcdb30 [rwRW,0x0000007ffffcdb30-0x0000007ffffcdb40]
$c17: 0x0000000040b94470 [rxR,0x0000000040b78000-0x0000000040bc9a00] (sentry)
$c18: 0x0000000041c8a500 [rwRW,0x0000000041c8a500-0x0000000041c8a5f0]
$c19: 0x0000000041c82000 [rwRW,0x0000000041c82000-0x0000000041c820d0]
$c20: 0x0000000040bc8700 [rxR,0x0000000040b78000-0x0000000040bc9a00]
$c21: 0x0000000041c8a400 [rwRW,0x0000000041c8a400-0x0000000041c8a4f0]
$c22: 0x0000000040bc8700 [rxR,0x0000000040b78000-0x0000000040bc9a00]
$c23: 0x0000007ffffdfca0 [rwRW,0x0000007ffffdfca0-0x0000007ffffdfcb0]
$c24: 0x0000007ffffda440 [rwRW,0x0000007ffbfe0000-0x0000007ffffe0000]
$c25: 0x0000000000000000
$c26: 0x0000000040726910 [rR,0x0000000040726900-0x000000004073bd80]
$c27: 0x0000000040ba4638 [rxR,0x0000000040b78000-0x0000000040bc9a00]
$c28: 0x0000000000000000
$c29: 0x0000000000000000
$c30: 0x0000000000000000
$c31: 0x0000000000000000
$pcc: 0x0000000040b93a74 [rxR,0x0000000040b78000-0x0000000040bc9a00]
Feb 22 14:24:18 cheribsd-mips64-hybrid kernel: USER_CHERI_EXCEPTION: pid 723 tid 100053 (func-pagemap-1), uid 0: CP2 fault (type 0x32)
Feb 22 14:24:18 cheribsd-mips64-hybrid kernel: Process arguments: /mnt/snmalloc-mips64-purecap-build/func-pagemap-1

Program received signal SIGPROT, CHERI protection violation
Capability tag faultwarning: GDB can't find the start of the function at 0x40b93a74.

    GDB is unable to find the start of the function at 0x40b93a74
and thus can't determine the size of that function's stack frame.
This means that GDB may be unable to access that stack frame, or
the frames below it.
    This problem is most likely caused by an invalid program counter or
stack pointer.
    However, if you think GDB should simply search farther back
from 0x40b93a74 for code which looks like the beginning of a
function, you can increase the range of the search using the `set
heuristic-fence-post' command.
 caused by register c1: 0x0000000041c8a100 [rwRW,0x41c8a200-0x41c8a2f0].
0x0000000040b93a74 in ?? ()
Reading symbols from /usr/libcheri/libexecinfo.so.1...
Reading symbols from /usr/libcheri/libc++.so.1...
Reading symbols from /usr/libcheri/libcxxrt.so.1...
Reading symbols from /usr/libcheri/libm.so.5...
Reading symbols from /usr/libcheri/libthr.so.3...
Reading symbols from /usr/libcheri/libc.so.7...
Reading symbols from /usr/libcheri/libelf.so.2...
Reading symbols from /usr/libcheri/libgcc_s.so.1...
Reading symbols from /libexec/ld-cheri-elf.so.1...

Thread 1 (LWP 100053 of process 723):
#0  0x0000000040b93a74 in scntree_RB_INSERT_COLOR (head=0x41c82080 [rwRW,0x41c82000-0x41c820d0], elm=0x41c8a300 [rwRW,0x41c8a300-0x41c8a3f0]) at /cheri/source/mainline/cheribsd/contrib/elftoolchain/libelf/elf_scn.c:52
#1  0x0000000040b94470 in scntree_RB_INSERT (head=0x41c82080 [rwRW,0x41c82000-0x41c820d0], elm=0x41c8a300 [rwRW,0x41c8a300-0x41c8a3f0]) at /cheri/source/mainline/cheribsd/contrib/elftoolchain/libelf/elf_scn.c:52
#2  0x0000000040b9abb8 in _libelf_allocate_scn (e=0x41c82000 [rwRW,0x41c82000-0x41c820d0], ndx=5) at /cheri/source/mainline/cheribsd/contrib/elftoolchain/libelf/libelf_allocate.c:162
#3  0x0000000040b94d44 in _libelf_load_section_headers (e=0x41c82000 [rwRW,0x41c82000-0x41c820d0], ehdr=<optimized out>) at /cheri/source/mainline/cheribsd/contrib/elftoolchain/libelf/elf_scn.c:121
#4  0x0000000040b95288 in elf_getscn (e=0x41c82000 [rwRW,0x41c82000-0x41c820d0], index=1) at /cheri/source/mainline/cheribsd/contrib/elftoolchain/libelf/elf_scn.c:162
#5  elf_nextscn (e=0x41c82000 [rwRW,0x41c82000-0x41c820d0], s=<optimized out>) at /cheri/source/mainline/cheribsd/contrib/elftoolchain/libelf/elf_scn.c:253
#6  0x00000000401e31dc in symtab_create (fd=3, bind=-1, type=2) at /cheri/source/mainline/cheribsd/contrib/libexecinfo/symtab.c:116
#7  0x00000000401e2860 in backtrace_symbols_fmt (trace=0x7ffffdade0 [rwRW,0x7ffffdade0-0x7ffffdede0], len=9, fmt=0x401d1f24 [rR,0x401d1f24-0x401d1f34] "%a <%n%D> at %f") at /cheri/source/mainline/cheribsd/contrib/libexecinfo/backtrace.c:197
#8  0x00000000401e2e44 in backtrace_symbols_fd_fmt (trace=0x41c82080 [rwRW,0x41c82000-0x41c820d0], len=9, fd=1, fmt=0x41c8a300 [rwRW,0x41c8a300-0x41c8a3f0] "") at /cheri/source/mainline/cheribsd/contrib/libexecinfo/backtrace.c:238
#9  backtrace_symbols_fd (trace=0x41c82080 [rwRW,0x41c82000-0x41c820d0], len=9, fd=1) at /cheri/source/mainline/cheribsd/contrib/libexecinfo/backtrace.c:259
[... snip snmalloc internals probably not germane...]
```

arichardson, 2021-03-18:
I haven't run the libunwind tests on RISC-V for a while. It's possible that they are currently broken.

arichardson, 2021-03-18:
Ah nevermind, I see the crash is inside libelf trying to read the symbol table. I wouldn't be surprised if libelf has lots of memory safety issues.

jrtc27, 2021-10-18:
Ah yeah, this is a repeat of a similar thing I saw in LLVM's stack trace implementation