CTSRD-CHERI/cheribsd #1005: Installing packages

abukharmeh, 2021-05-25:
Hi 

I am wondering how one is expected to install packages for Cheri BSD ? Is there any mirror with precompiled packages for  pkg manager or do I have to compile any package needed. pkg currently outputs 
`Bootstrapping pkg from pkg+http://pkg.FreeBSD.org/FreeBSD:13:riscv64c/latest, please wait...` but it looks like that mirror is down ! 

It looks like there is a script called /usr/bin/gdb-run.sh, but gdb is not installed nor gcc or llvm ?  

Thanks, 
Ibrahim

bsdjhb, 2021-06-03:
We do not currently provide any pre-compiled packages for CheriBSD.   It is possible to use FreeBSD precompiled packages (but they will be hybrid binaries and not CHERI).  However, FreeBSD does not yet provide precompiled RISC-V packages.

You can build gdb via cheribuild (cheribuild gdb-riscv64-hybrid-for-purecap-rootfs).  For building new binaries, you will likely want to cross-compile them from another host.  The simplest way to do that is to use the 'ccc' and 'cc++' wrappers from the cheri-exercises repo as CC and CXX (e.g. 'make CC="ccc riscv64-purecap"').

arichardson, 2021-06-03:
If you use cheribuild to compile GDB, it will be included in the disk image the next time you rebuild it.
cheribuild also installs compiler wrapper scripts that pass the required flags in `output/sdk/utils`, so if you haven't cloned the cheri-exercises you can also use
`make CC="$HOME/cheri/output/sdk/utils/cheribsd-riscv64-purecap-clang" CXX="$HOME/cheri/output/sdk/utils/cheribsd-riscv64-purecap-clang++"`

jrtc27, 2021-06-03:
Do note though that both of these are for personal use only; they are not stable interfaces and should not be embedded in build scripts (there is a reason ccc/cc++ is a helper script in cheri-exercises and those cheribsd-*-clang(++) scripts are in utils/, rather than either or both being in bin/). Build scripts should pass all the right flags to Clang manually.

abukharmeh, 2021-06-07:
Thank you all very much, yes I have seen CCC. Its nice to know if I used Cheri Build to build it, it will be automatically included next time I generate the image. 

While I was using EE to edit some config file, it crashed with a capability pointer something error. When I see such crashes, how do you expect me to report them, just the the actual error, or is there any certain log that would be helpful to attach alongside ?

arichardson, 2021-06-07:
@abukharmeh Details on what went wrong and instructions on how to reproduce the error are generally sufficient. If you have gdb installed, then re-running the command with `gdb-run.sh` and adding the backtrace to the bugreport would also be helpful.

closed 2021-06-07 by abukharmeh