CTSRD-CHERI/cheribsd #1009: VM map vs. large pages

nwf, 2021-05-28:
_Something_ seems to be going off the rails, though what, exactly, and whose fault it is I'm not sure.  Still, here's a good place for everything I know about it.

Setup: `omnetpp` with `snmalloc` and `mrs` loaded (but I don't think those matter) on the `caprevoke` branch at 662a2fa2b.

The kernel panics with `vm_caprevoke post past entry end: 51440000 > 51600000 (was 51400000)`.  That is, we began walking the page at `51400000` (hex) as part of a `vm_map_entry_t` that ended at `51440000` and that page carried us all the way to `51600000`.  That is, it's apparently large page that extends past the end of this `vm_map_entry_t`.

Looking around in the VM, we see the following mappings in increasing address order:

```
{left = ..., right = ..., start = 0x51440000, end = 0x5148f000, reservation = 0x51440000, next_read = 0x51440000, max_free = 0x0, object = {vm_object = 0xffffffd002437440, sub_map = ...}, offset = 0x9e000, eflags = 0x0, protection = 0x1b, max_protection = 0x1b, inheritance = 0x1, read_ahead = 0xf, wired_count = 0x0, cred = 0x0, wiring_thread = 0x0} 

{left = ..., right = ..., start = 0x513f1000, end = 0x51440000, reservation = 0x513f1000, next_read = 0x513f1000, max_free = 0x3e7ed80000, object = {vm_object = 0xffffffd002437440, sub_map = ...}, offset = 0x4f000, eflags = 0x0, protection = 0x1b, max_protection = 0x1b, inheritance = 0x1, read_ahead = 0xf, wired_count = 0x0, cred = 0x0, wiring_thread = 0x0}

{left = ..., right = ..., start = 0x514de000, end = 0x5152d000, reservation = 0x514de000, next_read = 0x514de000, max_free = 0x3e7ed80000, object = {vm_object = 0xffffffd002437440, sub_map = ...}, offset = 0x13c000, eflags = 0x0, protection = 0x1b, max_protection = 0x1b, inheritance = 0x1, read_ahead = 0xf, wired_count = 0x0, cred = 0x0, wiring_thread = 0x0}  
```

All of these are pointing at the same `vm_object_t`, namely

```
{lock = {lock_object = {lo_name = 0x25630000ffffffc00059c1c4, lo_flags = 0x25630000, lo_data = 0x0, lo_witness = 0xffffffd07fd8d700}, rw_lock = 0xffffffc0170db880}, object_list = {tqe_next = 0xffffffd002437550, tqe_prev = 0xffffffd002437350}, shadow_head = {lh_first = 0x0}, shadow_list = {le_next = 0x0, le_prev = 0x0}, memq = {tqh_first = 0xffffffd07cd9b590, tqh_last = 0xffffffd07cf7f660}, rtree = {rt_root = 0xffffffd00246bc60}, size = 0x452, domain = {dr_policy = 0x0, dr_iter = 0x0}, generation = 0x1, cleangeneration = 0x1, ref_count = 0xe, shadow_count = 0x0, memattr = 0x2, type = 0x0, pg_color = 0x13a2, flags = 0x13010, paging_in_progress = {__count = 0x0}, busy = {__count = 0x0}, resident_page_count = 0x434, backing_object = 0x0, backing_object_offset = 0x0, pager_object_list = {tqe_next = 0x0, tqe_prev = 0x0}, rvq = {lh_first = 0xffffffd07caf35c0}, handle = 0x0, un_pager = {vnp = {vnp_size = 0x0, writemappings = 0x0}, devp = {devp_pglist = {tqh_first = 0x0, tqh_last = 0x0}, ops = 0x0, dev = 0x0}, sgp = {sgp_pglist = {tqh_first = 0x0, tqh_last = 0x0}}, swp = {swp_tmpfs = 0x0, swp_blks = {pt_root = 0x0}, writemappings = 0x0}, phys = {ops = 0x0, {data_ptr = 0x0, data_val = 0x0}}}, cred = 0xffffffd00111ad00, charge = 0x452000, umtx_data = 0xc10000000000000000}
```

All three of those entries overlap the large page from `[51400000, 51600000)`, **but** there's a *gap* from `[5148f000, 514de000)` that nevertheless has a mapped page underneath it.  Something has gone wrong.

Moreover, that VM object is flagged as `0x13010 == OBJ_HASCAPS | OBJ_ONEMAPPING | OBJ_COLORED | OBJ_ANON`; I think the `OBJ_ONEMAPPING` is incorrect, as it has no fewer than three entries all pointing to it, but maybe anonymous mappings' `ONEMAPPING` means something subtly different.

bsdjhb, 2021-06-02:
So MAP_ANON might be quite special in that we do some things to try to permit future MAP_ANON mappings to share the same underlying VM object.  Certainly the multiple mappings sharing the same object is why you have multiple references, but it is still OBJ_ONEMAPPING.  You mentioned an assertion fails, can you give the file/line of the assertion that fails and maybe the stack trace?

nwf, 2021-06-02:
The particular assertion that trips is
https://github.com/CTSRD-CHERI/cheribsd/blob/662a2fa2b42471be7a851223221df5c36d74c21c/sys/vm/vm_caprevoke.c#L667-L669

The flow is a little convoluted, perhaps, but: at that line, `*addr` has been updated to point to the virtual address one past the end of the last page we walked in `vm_caprevoke_object_at` (and `oaddr` is what `*addr` was prior to that call).  The expectation, which may no longer be true (and may not have been at the time the code was written, even; I may have just been lucky to avoid tickling it so far) was that map entry boundaries always corresponded to page boundaries.  The backtrace at the time of the assertion trip isn't interesting: we're just running the MI bits of the revoker and the VM map state is what it is.

I'm happy to just remove the assertion if it's wrong or unnecessary, but what's worrying, and why I didn't rush to just excise it, is that, in this particular case, there's a gap in the map entries that still has a valid mapping underneath it, which made me worried that something more subtle is wrong.

bsdjhb, 2021-06-02:
Ah, I thought you had said it was an existing assertion in FreeBSD that was triggered which had puzzled me.  It may be though that we should add some new assertions to see if we can capture this case in stock FreeBSD.  I do wonder if it might have to do with growing the MAP_ANON object as the heap grows.

nwf, 2021-06-03:
This looks to be my misunderstanding of `psind`, not an actual large page in place.  Thanks, @bsdjhb.

closed 2021-06-03 by nwf