CTSRD-CHERI/cheribsd #1016: Morello /bin/csh loss of $csp tag

nwf, 2021-06-10:
I attempted to interrupt `gdb`'s `bt` with `^C` and grew impatient and hit `^Z`.  Instead of a shell prompt, I instead got a login prompt and a `csh.core`.  `gdb` says:

```
Core was generated by `-csh'.
Program terminated with signal SIGPROT, CHERI protection violation
Capability tag fault.
[...]
(gdb) bt
#0  process (catch=<optimized out>) at /home/nwf/cheri/cheribsd/contrib/tcsh/sh.c:1997
#1  0x00000000001322e0 in main (argc=<optimized out>, argv=<optimized out>)
    at /home/nwf/cheri/cheribsd/contrib/tcsh/sh.c:1430
(gdb) x/i $pc
=> 0x1331cc <process+1048>: ldr     w8, [sp, #1188]
(gdb) i r
[...]
csp            0xdc5d40003ffdbfff0000fffffff7f0c0 0xfffffff7f0c0 [rwRW,0xffffbff80000-0xfffffff80000]
[...]
tag_map        0x40007e0f8
```

That's https://github.com/CTSRD-CHERI/cheribsd/blob/47aecc899648a1f6fdea1474a967e5e57eb3bcb0/contrib/tcsh/sh.c#L1997 and `gdb` says we're trying to read `didexitset` and that it's `1` if we could, if that's informative.  This is 47aecc899648a1f6fdea1474a967e5e57eb3bcb0 with a80ab6773b060de0616241beaa02bb7d0a04f6c3 cherry-picked atop it, running on the FVP.

jrtc27, 2022-01-21:
This is a repeat of #198; a similar fix is needed for Morello and CHERI-RISC-V, but simplified as it's only needed for purecap and we don't need a union, we can just change the array element type to be a capability (and be sure to scale values appropriately)

bsdjhb, 2022-03-02:
Fixed for Morello in 3fda59c5a255c2fa92fe73c631e7e2150c78ae21 and RISC-V in 2fbaabfc23d8acd8dd0c86ed10cd559b483fcb9a.

closed 2022-03-02 by bsdjhb