CTSRD-CHERI/cheribsd #1045: Crash while running buggy IPC test

rwatson, 2021-07-13:
I introduced a bug in ```cheribsdtest-purecap-dynamic``` in which a process wedged itself performing an oversize IPC over a pipe. After hitting Ctrl-T, I hit Ctrl-C, leading to the following:
```
mi_switch+0x180 sleepq_wait+0x154 sleepq_wait_sig+0x2ca sleepq_wait_sig+0xc _sleep+0x25a sys_kbounce+0x115a user_writev+0x11e kern_write+0x8c sys_write+0x16 do_trap_user+0x3a0 cpu_exception_handler_user+0xde 
^Cra = 0xffffffc00053b76a
sp = 0xffffffc03224c7b0 [rwxRW,0x0000000000000000-0xffffffffffffffff]
gp = 0xffffffc000787f80 [rwxRW,0xffffffc000786884-0xffffffc000786c7a] (invalid,sealed)
tp = 0xffffffc0005ca6ff [rwxRW,0xffffffc0005ca884-0xffffffc0005cac7a] (invalid,sealed)
t[0] = 0x0000000000000001
t[1] = 0x0000000000010000
t[2] = 0x0000000000006000
t[3] = 0x0000000000000000
t[4] = 0x0000000000000000
t[5] = 0xffffffc03223d3a0
t[6] = 0x0000000000000001
s[0] = 0xffffffc03224c7c0
s[1] = 0xffffffc017a6a000
s[2] = 0x0000000000002000
s[3] = 0x0000000000000000
s[4] = 0x0000000000001000
s[5] = 0x0000000000001000
s[6] = 0xffffffd002144a60
s[7] = 0x0000000000000000
s[8] = 0xffffffd05a036000
s[9] = 0xffffffc03224c890
s[10] = 0x0000000000000fff
s[11] = 0x0000000000001000
a[0] = 0xffffffc017a6a000
a[1] = 0xffffffd05a036000
a[2] = 0x0000000000001000
a[3] = 0x00000000001c1924 [rwRW,0x00000000001c1924-0x00000000001c1928]
a[4] = 0xffffffc017a6a000
a[5] = 0x0000000000000100
a[6] = 0x0000000000000000
a[7] = 0x0000000000010000
sepc = 0xffffffc00053b872 [rwxRW,0x0000000000000000-0xffffffffffffffff]
ddc = 0x0000000000000000 [rwxRW,0x0000000000000000-0xffffffffffffffff]
sstatus == 0x0000000200004120
stval == 0xffffffc017a6a000
panic: Fatal capability page fault 0xffffffc00053b872: 0xffffffc017a6a000
cpuid = 0
time = 1626208435
KDB: stack backtrace:
db_trace_self() at db_trace_self
db_fetch_ksymtab() at db_fetch_ksymtab+0x170
kdb_backtrace() at kdb_backtrace+0x2c
vpanic() at vpanic+0x154
panic() at panic+0x52
do_trap_supervisor() at do_trap_supervisor+0x12a
cpu_exception_handler_supervisor() at cpu_exception_handler_supervisor+0xe2
--- exception 27, tval = 0xffffffc017a6a000
memmove() at memmove+0x11e
memmove() at memmove+0x12
uiomove_fromphys() at uiomove_fromphys+0x13a
sys_kbounce() at sys_kbounce+0x1296
user_writev() at user_writev+0x11e
kern_write() at kern_write+0x8c
sys_write() at sys_write+0x16
do_trap_user() at do_trap_user+0x3a0
cpu_exception_handler_user() at cpu_exception_handler_user+0xde
--- exception 8, tval = 0
KDB: enter: panic
[ thread pid 872 tid 100056 ]
Stopped at      kdb_enter+0x4c: sd      zero,0(a0)
db> show registers
ra          0xffffffc0002f72a2  kdb_enter+0x40
sp          0xffffffc03224c490
gp          0xffffffd05a036000
tp                        0x20
t0          0x8080808080808080
t1                           0
t2                           0
t3                           0
t4                           0
t5                           0
t6                           0
s0          0xffffffc03224c4b0
s1          0xffffffc0005d09f2
s2          0xffffffc03224c528
s3          0xffffffc0005bcfe2
s4          0xffffffc0171f2880
s5                       0x100
s6                         0x1
s7          0xffffffc00076bf10  sfstat+0x310
s8          0xffffffd05a036000
s9          0xffffffc03224c890
s10                      0xfff
s11                     0x1000
a0          0xffffffc000735fa0  kdb_why
a1          0x6b882345731d17d5
a2          0xffffffc03224c390
a3                         0x1
a4                         0x1
a5          0xffffffc00053c0d2  hybridabi_thread_setregs+0x166
a6                           0
a7          0xfefefefefefefeff
sepc        0xffffffc0002f72ae  kdb_enter+0x4c
sstatus            0x200004100
stval                        0
scause                     0x3
kdb_enter+0x4c: sd      zero,0(a0)
```

rwatson, 2021-07-13:
CheriBSD baseline for my branch is 8b75e190669e8d67c5cfdfd1627730aa7cda4239; no local kernel modifications.

rwatson, 2021-07-13:
Test in https://github.com/CTSRD-CHERI/cheribsd/pull/1027; 36489f16226a6b788d26f97ecff96894aef610a1 is the revision introducing a version of ```test_ipc_pipe_8k_tagstripped``` that reproducibly triggers the crash (hybrid kernel, ```cheribsdtest-purecap-dynamic``` binary).

brooksdavis, 2021-07-13:
It looks like either the stack is corrupt or the unwinder is confused, this isn't a sensible sequence of calls:

```
memmove() at memmove+0x11e
memmove() at memmove+0x12
uiomove_fromphys() at uiomove_fromphys+0x13a
sys_kbounce() at sys_kbounce+0x1296
user_writev() at user_writev+0x11e
kern_write() at kern_write+0x8c
sys_write() at sys_write+0x16
do_trap_user() at do_trap_user+0x3a0
cpu_exception_handler_user() at cpu_exception_handler_user+0xde
```

jrtc27, 2021-07-13:
That's memmove being a strong alias for memcpy, with memcpy calling _memcpy, a static function whose symbol won't be in the non-full kernel (but whose code is emitted just after memcpy, i.e. memmove), showing up.
```
ffffffc00053bcc8 <memmove>:
ffffffc00053bcc8: 01 11 addi    sp, sp, -32
ffffffc00053bcca: 06 ec sd      ra, 24(sp)
ffffffc00053bccc: 22 e8 sd      s0, 16(sp)
ffffffc00053bcce: 26 e4 sd      s1, 8(sp)
ffffffc00053bcd0: 00 10 addi    s0, sp, 32
ffffffc00053bcd2: aa 84 mv      s1, a0
ffffffc00053bcd4: 85 46 addi    a3, zero, 1
ffffffc00053bcd6: 97 00 00 00   auipc   ra, 0
ffffffc00053bcda: e7 80 40 01   jalr    20(ra)
ffffffc00053bcde: 26 85 mv      a0, s1
ffffffc00053bce0: a2 64 ld      s1, 8(sp)
ffffffc00053bce2: 42 64 ld      s0, 16(sp)
ffffffc00053bce4: e2 60 ld      ra, 24(sp)
ffffffc00053bce6: 05 61 addi    sp, sp, 32
ffffffc00053bce8: 82 80 ret

ffffffc00053bcea <_memcpy>:
```

jrtc27, 2021-07-13:
(and that sys/sys/systm.h defines bcopy as __builtin_memmove)

rwatson, 2021-07-14:
@jrtc27: I think it’s the path from ```sys_writeve(2)``` into ```sys_kbounce()``` that is unexpected. I’ll create a new test case that it specifically about this bug, using a timer rather than a user hitting Ctrl-C.

rwatson, 2021-07-14:
Reproduced deterministically by a new dedicated test #1046.

rwatson, 2021-07-17:
Huh:

```
(gdb) l *sys_kbounce+0x1296
0xffffffc00031f84a is in pipe_write (/usr/home/robert/cheri/cheribsd/sys/kern/sys_pipe.c:969).
```

That makes a lot more sense.

commit 22af6259 (2021-07-18): Add a uiomove_fromphys_cap and make uiomove_fromphys strip tags

This mirrors the normal uiomove where uiomove is tag-stripping and
uiomove_cap preserves tags. Currently pipe_clone_write_buffer uses
uiomove_fromphys to copy the userspace buffer to the kernel buffer and
so bypasses the userspace capability's lack of LOAD_CAP permission. This
means that if the userspace buffer contains valid capabilities, the
in-kernel bcopy will end up loading the valid capabilities rather than
stripping their tags, and storing them to the kernel buffer will take a
store capability page fault since the pages we [... cut]
commit 72ca6814 (2021-08-04): Add a uiomove_fromphys_cap and make uiomove_fromphys strip tags

This mirrors the normal uiomove where uiomove is tag-stripping and
uiomove_cap preserves tags. Currently pipe_clone_write_buffer uses
uiomove_fromphys to copy the userspace buffer to the kernel buffer and
so bypasses the userspace capability's lack of LOAD_CAP permission. This
means that if the userspace buffer contains valid capabilities, the
in-kernel bcopy will end up loading the valid capabilities rather than
stripping their tags, and storing them to the kernel buffer will take a
store capability page fault since the pages we [... cut]
closed 2021-08-04 by jrtc27