CTSRD-CHERI/cheribsd #1058: panic: vm_reserv_from_object: msucc doesn't succeed pindex

arichardson, 2021-07-29:
Just saw the following panic while running GDB on CHERI-RISC-V (hybrid kernel)
```
panic: vm_reserv_from_object: msucc doesn't succeed pindex
cpuid = 0
time = 1627576392
KDB: stack backtrace:
db_trace_self() at db_trace_self
db_fetch_ksymtab() at db_fetch_ksymtab+0x170
kdb_backtrace() at kdb_backtrace+0x2c
vpanic() at vpanic+0x154
panic() at panic+0x52
vm_reserv_alloc_contig() at vm_reserv_alloc_contig+0x6be
vm_reserv_alloc_page() at vm_reserv_alloc_page+0x62
vm_page_alloc_domain_after() at vm_page_alloc_domain_after+0x15a
vm_page_grab_pages() at vm_page_grab_pages+0x202
vm_page_grab_pages_unlocked() at vm_page_grab_pages_unlocked+0x1be
allocbuf() at allocbuf+0x2e0
getblkx() at getblkx+0x6b2
cluster_read() at cluster_read+0xb8
ffs_own_mount() at ffs_own_mount+0x34a0
VOP_READ_APV() at VOP_READ_APV+0x20
vn_lock_pair() at vn_lock_pair+0x57e
vn_lock_pair() at vn_lock_pair+0x28e
vn_rdwr() at vn_rdwr+0x462
user_copy_file_range() at user_copy_file_range+0x28e
user_readv() at user_readv+0x120
user_read() at user_read+0x8c
freebsd64_read() at freebsd64_read+0x66
do_trap_user() at do_trap_user+0x3a0
cpu_exception_handler_user() at cpu_exception_handler_user+0xde
--- exception 8, tval = 0
KDB: enter: panic
[ thread pid 780 tid 100064 ]
Stopped at      kdb_enter+0x4c: sd      zero,0(a0)
```

markjdb, 2024-12-21:
This is probably OBE? In the intervening years there's been a number of riscv pmap bug fixes which might have addressed this.

bsdjhb, 2024-12-30:
Let's just close this.  If it happens again it can be reopened.

closed 2024-12-30 by bsdjhb