CTSRD-CHERI/cheribsd #1062: Rename CHERI_PERM_CHERIABI_VMMAP

brooksdavis, 2021-08-10:
This permission applies to both the mmap family of APIs and shmdt so VMMAP is not the right name.  Additionally, the permission isn't strictly tied to CheriABI (in a hybrid world were not all of VA is part of DDC, we would want to use this permission for the APIs that manage VA outside DDC.)

The proposed replacement is CHERI_PERM_VMMAN.

bsdjhb, 2021-08-10:
Superhero connotations are ok with me.  If you wanted something shorter you could perhaps just use `CHERI_PERM_VM`.  One observation is that VM is now an overloaded term in systems software (virtual memory vs virtual machines) so `CHERI_PERM_VMEM` might be less ambiguous.

kevin-brodsky-arm, 2021-08-11:
VMMAN is supposed to be a shorthand for Virtual Memory Management, but I do see the superhero connotation now! VMEM would be fine as well I suppose.

One aspect that I find potentially confusing with `CHERI_PERM_VMMAN` or similar is that it doesn't distinguish that software-defined permission from the hardware-defined permissions. Maybe `CHERI_PERM_SW_VMMAN` would work?

brooksdavis, 2022-01-25:
Further discussion has us leaning toward `CHERI_PERM_VMEM` or `CHERI_PERM_SW_VMEM` with the non-code spelling being "VMem"

brooksdavis, 2022-01-25:
After further discussion we'll go with `CHERI_PERM_SW_VMEM`

PR #1269 (2022-01-25): CHERIABI_VMMAP -> SW_VMEM
Rename to CHERI_PERM_SW_VMEM and make CHERI_PERM_CHERIABI_VMMAP expand
to include a warning to give out-of-tree users a chance to migrate
before removing the old define.

This permission applies to both the mmap family of APIs and shmdt so
VMMAP is not the right name. Additionally, the permission isn't strictly
tied to CheriABI (in a hybrid world were not all of VA is part of DDC,
we would want to use this permission for the APIs that manage VA outside
DDC.)

Resolves #1062
commit 00d71bd4 (2022-01-25): CHERIABI_VMMAP -> SW_VMEM

Rename to CHERI_PERM_SW_VMEM and make CHERI_PERM_CHERIABI_VMMAP expand
to include a warning to give out-of-tree users a chance to migrate
before removing the old define.

This permission applies to both the mmap family of APIs and shmdt so
VMMAP is not the right name. Additionally, the permission isn't strictly
tied to CheriABI (in a hybrid world were not all of VA is part of DDC,
we would want to use this permission for the APIs that manage VA outside
DDC.)

Resolves #1062
closed 2022-01-26 by brooksdavis