CTSRD-CHERI/cheribsd #1249: Switch to new Morello varargs ABI

jrtc27, 2022-01-19:
Will need:
* Passing `-Xclang -morello-vararg=new`
* Updating syscall(2) implementation

Toolchain support merged to morello/dev on 10th December, so likely not able to assume its existence quite yet in users' existing builds.

bsdjhb, 2022-03-09:
We probably want to do this before the first release so we don't have to deal with a flag day for this later?

jrtc27, 2022-03-09:
Yeah... fun for native builds I imagine

bsdjhb, 2022-03-17:
Looks like the syscall() handling in the kernel already assumes all the arguments are only in registers?

jrtc27, 2022-03-17:
Oh right, yeah, because the whole point is it _does_ use registers and that doesn't catch overflow, the new ABI is what shoves everything on the stack and creates a bounded capability

kwitaszczyk, 2022-05-25:
The new var-args ABI was merged to `dev` and released in 22.05. CheriABI packages and hybrid ABI packages were rebuilt with the new ABI and verified.

closed 2022-05-25 by kwitaszczyk