CTSRD-CHERI/cheribsd #1284: Morello virtio rand panic in qemu

nwf, 2022-02-02:
I left a Morello qemu instance running idle for a while and came back to discover it dead with
```
x0: 0x0000000000000000
  x1: 0xffff0000a1c5c080 (__bss_end + a0c16ec0)
  x2: 0x0000000000000010
  x3: 0x0000000000000000
  x4: 0xffff00000076aeb0 (generic_bs_w_2 + 0)
  x5: 0x0000000000000000 [rwxRW,0x0000000000000000-0xffffffffffffffff]
  x6: 0x0000000000000000 [,0x0000000000000007-0x0000000000000007] (invalid)
  x7: 0x0000000000000100
  x8: 0xffffa000006cad00
  x9: 0xffff000000bf7b98 (memmap_bus + 0)
 x10: 0xffff000000b8e560 (virtio_bus_notify_vq_desc + 0)
 x11: 0x0000000000000abc
 x12: 0x0000000000000004
 x13: 0xffff0000dc734000 (__bss_end + db6eee40)
 x14: 0xffff0000a1c42710 (__bss_end + a0bfd550)
 x15: 0x0000000000000008
 x16: 0xffff0000e1021c50 (_DYNAMIC + 1b8)
 x17: 0xffff000000304968 (virtqueue_notify + 0)
 x18: 0xffff0000a1c426b0 (__bss_end + a0bfd4f0)
 x19: 0xffffa000008cc800
 x20: 0xffffa000004e3200
 x21: 0x0000000000000010
 x22: 0x0000000000000000
 x23: 0xffff000000d03598 (harvest_context + 0)
 x24: 0xffff0000002572cc (random_fortuna_process_event + 0)
 x25: 0x000000000000409c
 x26: 0x0000000000000029
 x27: 0xffff000000d07628 (harvest_context + 4090)
 x28: 0x0000000000000008
 x29: 0xffff0000a1c426b0 (__bss_end + a0bfd4f0)
 ddc: 0x0000000000000000
  sp: 0xffff0000a1c426b0
  lr: 0xffff000000304a14 (virtqueue_notify + ac)
 elr: 0xffff000000304a14 [rwxRW,0x0000000000000000-0xffffffffffffffff] (invalid) (virtqueue_notify + ac)
spsr:         60000045
 far:                0
 esr:         86000028
panic: Capability abort from kernel space: tag violation
cpuid = 0
time = 1643339834
KDB: stack backtrace:
db_trace_self() at db_trace_self
db_trace_self_wrapper() at db_trace_self_wrapper+0x30
vpanic() at vpanic+0x184
panic() at panic+0x70
cap_abort() at cap_abort+0x1d4
handle_el1h_sync() at handle_el1h_sync+0x7c
--- exception, esr 0x86000028
virtqueue_notify() at virtqueue_notify+0xac
vtrnd_read() at vtrnd_read+0x8c
random_kthread() at random_kthread+0x230
fork_exit() at fork_exit+0x74
fork_trampoline() at fork_trampoline+0x10
KDB: enter: panic
[ thread pid 5 tid 100039 ]
Stopped at      kdb_enter+0x44: undefined       f903011f
```

The kernel identifies itself as
```
FreeBSD 14.0-CURRENT #11 dev-n251073-e6d5cf5ee5e8-dirty: Fri Jan 28 01:12:10 UTC 2022
    daemon@afd5f6c4e6fd:/cheri/build/mainline/cheribsd-morello-purecap-build/cheri/source/mainline/cheribsd/arm64.aarch64c/sys/GENERIC-MORELLO arm64
clang version 11.0.0 (https://git.morello-project.org/morello/llvm-project.git 94e1dbacf1d854b48386ec2c07a35e0694d626e2)
```
and `e6d5cf5ee5e8` is some local commits that shouldn't be relevant atop f49d94c1159b18f983274fbf08c5b889a670beea.

arichardson, 2022-02-02:
Is Morello QEMU up-to-date? Does it include https://github.com/CTSRD-CHERI/qemu/commit/7867982da14516ff3ea803dfe62a242128a5d452?

jrtc27, 2022-02-02:
qemu-morello-merged has included that commit since 8th Nov, but if you haven't built morello-qemu since then you won't have the fix. I'm reminded I need to figure out how to gracefully migrate people from morello-qemu to qemu without giving strange/confusing errors all of a sudden...

nwf, 2022-02-02:
Hm!  This is probably another case of user error, but...
```
$ ls -l /cheri/out/mainline/sdk/bin/qemu-system-morello 
-rwxr-xr-x 1 daemon daemon 66470696 Jan  4 10:22 /cheri/out/mainline/sdk/bin/qemu-system-morello
```
certainly suggests I've built relatively recently, but
```
$ /cheri/out/mainline/morello-sdk/bin/qemu-system-morello --version
QEMU emulator version 5.2.0 (v2.4.0-44058-gfae78175f318)
```
suggests it's half a year out of date.  I guess the last time I built I had `--skip-update` passed or something.  Sorry for the noise.

closed 2022-02-02 by nwf