CTSRD-CHERI/cheribsd #1302: CHERI exception when restoring capabilities in the swap pager

markjdb, 2022-02-21:
I used the following test program to exercise capability tag save/restore in the swap pager:

```
#include <sys/mman.h>
#include <err.h>
#include <stdlib.h>
#include <string.h>

int
main(void)
{
        char *ptr;
        size_t sz;

        sz = 3 * 1024 * 1024 * 1024ul;
        ptr = mmap(NULL, sz, PROT_READ | PROT_WRITE, MAP_ANON, -1, 0);
        if (ptr == MAP_FAILED)
                err(1, "mmap");

        memset(ptr, 0, sz);
        memset(ptr, 0, sz);
}
```

The size is chosen to ensure that some fraction of the buffer will be paged out and later paged back in. On a riscv-purecap kernel, I reliably hit a "panic: CHERI exception 0x1 at 0xffffffc000713fda":

```
#12 <signal handler called>
#13 kdb_enter (why=<optimized out>, msg=<optimized out>) at /usr/home/markj/cheri/cheribsd/sys/kern/subr_kdb.c:506
#14 0xffffffc000417f7a in vpanic (fmt=0xffffffc000800f64 [rR,0xffffffc000800f64-0xffffffc000800f86] "CHERI exception %#lx at 0x%016lx\n", 
    ap=0xffffffc001b24b60 [rwxRW,0xffffffc001b21000-0xffffffc001b26000]) at /usr/home/markj/cheri/cheribsd/sys/kern/kern_shutdown.c:937
#15 0xffffffc000417c24 in panic (fmt=0xffffffc000800f64 [rR,0xffffffc000800f64-0xffffffc000800f86] "CHERI exception %#lx at 0x%016lx\n")
    at /usr/home/markj/cheri/cheribsd/sys/kern/kern_shutdown.c:874
#16 0xffffffc00078ff06 in do_trap_supervisor (frame=0xffffffc001b24bb0 [rwxRW,0xffffffc001b21000-0xffffffc001b26000])
    at /usr/home/markj/cheri/cheribsd/sys/riscv/riscv/trap.c:477
#17 <signal handler called>
#18 cheri_restore_tag (cp=<optimized out>) at /usr/home/markj/cheri/cheribsd/sys/vm/swap_pager.c:2224
#19 swp_pager_meta_cheri_get_tags (page=0xffffffd07fc609e0 [rwxRW,0xffffffd000000000-0xffffffd080000000]) at /usr/home/markj/cheri/cheribsd/sys/vm/swap_pager.c:2262
#20 swp_pager_async_iodone (bp=0xffffffc07f546240 [rwxRW,0xffffffc07f546240-0xffffffc07f547470]) at /usr/home/markj/cheri/cheribsd/sys/vm/swap_pager.c:1757
#21 0xffffffc0004fd4de in bufdone (bp=0xffffffc07f546240 [rwxRW,0xffffffc07f546240-0xffffffc07f547470]) at /usr/home/markj/cheri/cheribsd/sys/kern/vfs_bio.c:4564
#22 0xffffffc000714c50 in swapgeom_done (bp2=0xffffffd0041aa000 [rwxRW,0xffffffd0041aa000-0xffffffd0041aa240]) at /usr/home/markj/cheri/cheribsd/sys/vm/swap_pager.c:3027
#23 0xffffffc0004fd154 in biodone (bp=0xffffffd0041aa000 [rwxRW,0xffffffd0041aa000-0xffffffd0041aa240]) at /usr/home/markj/cheri/cheribsd/sys/kern/vfs_bio.c:4456
#24 0xffffffc0003355ca in g_io_deliver (bp=0xffffffd0041aa000 [rwxRW,0xffffffd0041aa000-0xffffffd0041aa240], error=0)
```

which points to the `cheri_seal()` call in `cheri_restore_tag()`.

bsdjhb, 2022-03-02:
Fixed in d2f3b99e6000626b3c924230f24d895ba9eee8a0.

bsdjhb, 2022-03-02:
Oops, my bad.

jrtc27, 2022-10-13:
https://github.com/CTSRD-CHERI/cheribsd/pull/1517 probably fixes whatever this "new problem that [you] don't really understand" was

markjdb, 2022-10-13:
> #1517 probably fixes whatever this "new problem that [you] don't really understand" was

That would be the panic I reported above.