CTSRD-CHERI/cheribsd #1313: Update portsnap(8) to use cheribsd-ports.

kwitaszczyk, 2022-03-08:
Currently, portsnap uses portsnap.FreeBSD.org to bootstrap /usr/ports. We should modify it to use cheribsd-ports instead.

brooksdavis, 2022-03-08:
For now we should just patch it to fail with a pointer to the cheribsd-port git repo if installed on a system where the default ABI is purecap. This does of course raise bootstrapping questions, but I think our answer is going to be packages.

jrtc27, 2022-03-08:
GitHub exports zip files, we could make it grab those

brooksdavis, 2022-03-08:
> GitHub exports zip files, we could make it grab those

That's plausible, although I'm not quite sure what the update path would look like. Probably nuke everything but /usr/ports/distfiles and extract?

bsdjhb, 2023-12-06:
Isn't portsnap OBE now?