CTSRD-CHERI/cheribsd #1321: Recent dev vs. tools/build/test-includes

nwf, 2022-03-17:
FWIW, a clean build of `dev` fails to make much progress on my machine because all the `sys/_*.h` files now get included and all, quite deliberately, cause the compiler to bomb out with "don't do that" errors.

I *believe* this an upstream FreeBSD issue and is, ultimately, because
https://github.com/CTSRD-CHERI/cheribsd/blob/d3bb19c14e9afcb987a911905a79b6a24b32d85e/tools/build/test-includes/Makefile#L19
is not actually spec conformant.  In particular, (X/Open CAE Specification Commands and Utilities Issue 4, Version 2)[https://pubs.opengroup.org/onlinepubs/009656399/toc.pdf] sayeth:

    The description of basic regular expression bracket expressions in the XBD specification, Section
    7.3.5, RE Bracket Expression also applies to the pattern bracket expression, except that the
    exclamation-mark character (!) replaces the circumflex character (ˆ) in its role in a non-matching
    list in the regular expression notation. A bracket expression starting with an unquoted
    circumflex character produces unspecified results.

There must be something about my setup that elicits these "unspecified results", though I hesitate to speculate which of its many, many "charming features", exactly, that might be.

brooksdavis, 2022-03-17:
From the FreeBSD manpage:
> The character class may be complemented by making an exclamation
> point (`!') the first character of the character class.  A caret (`^')
> has the same effect but is non-standard.

Could you test with `!` in place of `^`? If that works I can prod Warner.

nwf, 2022-03-17:
That (`!` and not `^`) seems to work fine.

brooksdavis, 2022-03-17:
Submitted for review https://reviews.freebsd.org/D34597

brooksdavis, 2022-04-26:
Merged to dev

closed 2022-04-26 by brooksdavis