CTSRD-CHERI/cheribsd #1324: FreeBSD boot loader banner should print “CheriBSD”

rwatson, 2022-03-21:
A minor issue of branding, although also useful for avoiding confusion -- it would be nice if the ASCII art ‘FreeBSD’ read ‘CheriBSD’ in the boot loader, to make it more obvious to users that that’s what they are using [and us, if debugging].

There are probably some other things like this, but we’ve caught quite a few of the obvious ones, such as /etc/motd.

jrtc27, 2022-03-21:
Do we also update all the references in the installer? Change the path in /boot/EFI to be cheribsd not freebsd, and the scripts (yes, multiple) that know about it? We're also (deliberately, because configure scripts) not changing uname's output. There's always going to be a _lot_ of FreeBSD branding that remains...

rwatson, 2022-03-21:
I have in mind only limited changes in branding -- and especially avoiding ones that would, for example, disrupt software porting efforts (such as uname).

closed 2022-04-08 by rwatson