CTSRD-CHERI/cheribsd #1335: Simplify bsdinstall installation for Morello boxes

rwatson, 2022-03-27:
There are a number of steps we could potentially take to reduce the degree of user interaction required to install CheriBSD, including:

- Default to xterm (and perhaps don't prompt at all)
- Sort a UK keyboard option to the top of the list
- Don't ask for a hostname
- Shorten the pure-capability kernel name so it fits in the dialog
- Don't actually ask about what distributions to install at all; especially don't ask about 'src' since we'd rather they checked it out explicitly using git and our supported development tooling
- Consider always using the entire disk
- If the entire disk is being used, simply select GPT automatically and don't offer bizarre other options
- Don't show the partition editor if they selected the entire disk?
- If possible, hide the mounted installation media (typically a USB disk) from the user when offering disks to install on, partitions to edit, and so on.

rwatson, 2022-03-27:
Further thought:
- The Keymap Selection should identify the default keycap in parenthesis in the menu option for using it, rather than hiding that in descriptive text.

jrtc27, 2022-03-27:
> * Don't ask for a hostname

This seems like it's asking for trouble...

> * Shorten the pure-capability kernel name so it fits in the dialog

Not a problem if the dialogue is made a non-stupid size... see https://reviews.freebsd.org/D34414

rwatson, 2022-03-27:
- If using DHCP, don't ask for further resolver configuration
- Skip the System Configuration menu
- Skip the System Hardening menu

rwatson, 2022-03-27:
> > * Don't ask for a hostname
> 
> This seems like it's asking for trouble...

True. We could default it to “cheribsd-morello” or something.
 
> > * Shorten the pure-capability kernel name so it fits in the dialog
> 
> Not a problem if the dialogue is made a non-stupid size... see https://reviews.freebsd.org/D34414

Does this work with live resizing when the user realises that the window is too small after starting the installation process and/or getting to that dialog? Or just if they have a big window to begin with...

rwatson, 2022-04-08:
Another possible thing:

- Don’t prompt for keyboard mapping selection on a serial console

rwatson, 2022-05-02:
Many of these concerns were addressed in #1345, but some remain and so I’ve left this issue open while we ponder them further.