CTSRD-CHERI/cheribsd #1336: Improve buildworld defaults when building natively

arichardson, 2022-03-27:
We currently have to manually set the following config options (see https://github.com/CTSRD-CHERI/cheripedia/wiki/HOWTO:-Build-CheriBSD-natively-on-Morello). It would be nice if `make buildworld` just worked.
```
WITH_CHERI=

WITHOUT_OFED=

WITHOUT_CLANG=
WITHOUT_LLD=
WITHOUT_LLDB=

WITHOUT_CLANG_BOOTSTRAP=
WITHOUT_LLD_BOOTSTRAP=

WITHOUT_MODULES=	armv8crypto linprocfs linux64 linux_common efirt ena mlx4 mlx4en mlx5 mlx5en
```

```
# make showconfig
make[1]: "/usr/src/share/mk/bsd.linker.mk" line 84: warning: /usr/src: Rerunning /opt/morello-hybrid/morello-llvm-project/bin/ld.lld -v to compute LINKER_TYPE/LINKER_VERSION. This value should be cached!
MK_ACCT          = yes
MK_ACPI          = yes
MK_APM           = yes
MK_ASAN          = no
MK_ASSERT_DEBUG  = yes
MK_AT            = yes
MK_ATM           = yes
MK_AUDIT         = yes
MK_AUTHPF        = yes
MK_AUTOFS        = yes
MK_AUTO_OBJ      = yes
MK_BEARSSL       = no
MK_BHYVE         = yes
MK_BHYVE_SNAPSHOT = no
MK_BIND_NOW      = no
MK_BLACKLIST     = yes
MK_BLACKLIST_SUPPORT = yes
MK_BLUETOOTH     = yes
MK_BOOT          = yes
MK_BOOTPARAMD    = yes
MK_BOOTPD        = yes
MK_BSDINSTALL    = yes
MK_BSD_CPIO      = yes
MK_BSNMP         = yes
MK_BZIP2         = yes
MK_BZIP2_SUPPORT = yes
MK_CALENDAR      = yes
MK_CAPSICUM      = yes
MK_CAROOT        = yes
MK_CASPER        = yes
MK_CCACHE_BUILD  = no
MK_CCD           = yes
MK_CDDL          = no
MK_CHERI         = no
MK_CHERIBSDBOX   = yes
MK_CLANG         = yes
MK_CLANG_BOOTSTRAP = no
MK_CLANG_EXTRAS  = no
MK_CLANG_FORMAT  = no
MK_CLANG_FULL    = yes
MK_CLANG_IS_CC   = yes
MK_CLEAN         = no
MK_COMPAT_CHERIABI = no
MK_CPP           = yes
MK_CROSS_COMPILER = yes
MK_CRYPT         = yes
MK_CTF           = no
MK_CUSE          = yes
MK_CXGBETOOL     = yes
MK_CXX           = yes
MK_DEBUG_FILES   = yes
MK_DETECT_TZ_CHANGES = no
MK_DIALOG        = yes
MK_DICT          = yes
MK_DIRDEPS_BUILD = no
MK_DIRDEPS_CACHE = no
MK_DMAGENT       = yes
MK_DOCCOMPRESS   = yes
MK_DTRACE_TESTS  = no
MK_DYNAMICROOT   = yes
MK_EE            = yes
MK_EFI           = yes
MK_ELFTOOLCHAIN_BOOTSTRAP = yes
MK_EXAMPLES      = yes
MK_EXPERIMENTAL  = no
MK_EXTRA_TCP_STACKS = no
MK_FDT           = yes
MK_FILE          = yes
MK_FINGER        = yes
MK_FLOPPY        = yes
MK_FORMAT_EXTENSIONS = yes
MK_FORTH         = yes
MK_FP_LIBC       = yes
MK_FREEBSD_UPDATE = yes
MK_FTP           = yes
MK_GAMES         = yes
MK_GH_BC         = yes
MK_GNU_DIFF      = yes
MK_GOOGLETEST    = yes
MK_GPIO          = yes
MK_GSSAPI        = yes
MK_HAST          = yes
MK_HESIOD        = no
MK_HTML          = yes
MK_HYPERV        = no
MK_ICONV         = yes
MK_INCLUDES      = yes
MK_INET          = yes
MK_INET6         = yes
MK_INET6_SUPPORT = yes
MK_INETD         = yes
MK_INET_SUPPORT  = yes
MK_INIT_ALL_PATTERN = no
MK_INIT_ALL_ZERO = no
MK_INSTALLLIB    = yes
MK_INSTALL_AS_USER = no
MK_IPFILTER      = yes
MK_IPFW          = yes
MK_IPSEC_SUPPORT = yes
MK_ISCSI         = yes
MK_JAIL          = yes
MK_KDUMP         = yes
MK_KERBEROS      = yes
MK_KERBEROS_SUPPORT = yes
MK_KERNEL_RETPOLINE = no
MK_KERNEL_SYMBOLS = yes
MK_KVM           = yes
MK_KVM_SUPPORT   = yes
MK_LDNS          = yes
MK_LDNS_UTILS    = yes
MK_LEGACY_CONSOLE = yes
MK_LIB32         = no
MK_LIB64         = yes
MK_LLD           = yes
MK_LLDB          = yes
MK_LLD_BOOTSTRAP = yes
MK_LLD_IS_LD     = yes
MK_LLVM_ASSERTIONS = yes
MK_LLVM_BINUTILS = no
MK_LLVM_COV      = yes
MK_LLVM_CXXFILT  = yes
MK_LLVM_TARGET_AARCH64 = yes
MK_LLVM_TARGET_ALL = yes
MK_LLVM_TARGET_ARM = yes
MK_LLVM_TARGET_BPF = no
MK_LLVM_TARGET_MIPS = no
MK_LLVM_TARGET_POWERPC = yes
MK_LLVM_TARGET_RISCV = yes
MK_LLVM_TARGET_X86 = yes
MK_LOADER_EFI_SECUREBOOT = no
MK_LOADER_FIREWIRE = no
MK_LOADER_GELI   = yes
MK_LOADER_KBOOT  = no
MK_LOADER_LUA    = yes
MK_LOADER_OFW    = no
MK_LOADER_UBOOT  = no
MK_LOADER_VERBOSE = no
MK_LOADER_VERIEXEC = no
MK_LOADER_VERIEXEC_PASS_MANIFEST = no
MK_LOADER_VERIEXEC_VECTX = no
MK_LOADER_ZFS    = no
MK_LOCALES       = yes
MK_LOCATE        = yes
MK_LPR           = yes
MK_LS_COLORS     = yes
MK_MAIL          = yes
MK_MAILWRAPPER   = yes
MK_MAKE          = yes
MK_MAKE_CHECK_USE_SANDBOX = yes
MK_MALLOC_PRODUCTION = no
MK_MAN           = yes
MK_MANCOMPRESS   = yes
MK_MANSPLITPKG   = no
MK_MAN_UTILS     = yes
MK_META_MODE     = no
MK_MLX5TOOL      = yes
MK_NDIS          = yes
MK_NETCAT        = yes
MK_NETGRAPH      = yes
MK_NETGRAPH_SUPPORT = yes
MK_NIS           = yes
MK_NLS           = yes
MK_NLS_CATALOGS  = yes
MK_NS_CACHING    = no
MK_NTP           = yes
MK_NVME          = yes
MK_OFED          = yes
MK_OFED_EXTRA    = no
MK_OPENLDAP      = no
MK_OPENMP        = no
MK_OPENSSH       = yes
MK_OPENSSL       = yes
MK_OPENSSL_KTLS  = yes
MK_PAM           = yes
MK_PAM_SUPPORT   = yes
MK_PF            = yes
MK_PIE           = yes
MK_PKGBOOTSTRAP  = yes
MK_PMC           = yes
MK_PORTSNAP      = yes
MK_PPP           = yes
MK_PROFILE       = no
MK_QUOTAS        = yes
MK_RADIUS_SUPPORT = yes
MK_RATELIMIT     = no
MK_RBOOTD        = yes
MK_REPRODUCIBLE_BUILD = no
MK_RESCUE        = yes
MK_RETPOLINE     = no
MK_ROUTED        = yes
MK_RPCBIND_WARMSTART_SUPPORT = no
MK_SCTP_SUPPORT  = yes
MK_SENDMAIL      = yes
MK_SERVICESDB    = yes
MK_SETUID_LOGIN  = yes
MK_SHAREDOCS     = yes
MK_SHARED_TOOLCHAIN = yes
MK_SORT_THREADS  = no
MK_SOURCELESS    = yes
MK_SOURCELESS_HOST = yes
MK_SOURCELESS_UCODE = yes
MK_SSP           = yes
MK_STAGING       = no
MK_STAGING_MAN   = no
MK_STAGING_PROG  = no
MK_STALE_STAGED  = no
MK_STATS         = yes
MK_SYSCONS       = yes
MK_SYSROOT       = no
MK_SYSTEM_COMPILER = yes
MK_SYSTEM_LINKER = yes
MK_TALK          = yes
MK_TCP_WRAPPERS  = yes
MK_TCSH          = yes
MK_TELNET        = yes
MK_TESTS         = yes
MK_TESTS_SUPPORT = yes
MK_TEXTPROC      = yes
MK_TFTP          = yes
MK_TOOLCHAIN     = yes
MK_UBSAN         = no
MK_UNBOUND       = yes
MK_UNIFIED_OBJDIR = yes
MK_USB           = yes
MK_USB_GADGET_EXAMPLES = yes
MK_UTMPX         = yes
MK_VERIEXEC      = no
MK_VI            = yes
MK_VT            = yes
MK_WARNS         = yes
MK_WERROR        = yes
MK_WIRELESS      = yes
MK_WIRELESS_SUPPORT = yes
MK_WPA_SUPPLICANT_EAPOL = yes
MK_ZFS           = no
MK_ZONEINFO      = yes
MK_ZONEINFO_LEAPSECONDS_SUPPORT = no
```

jrtc27, 2022-03-27:
efirt no longer needs disabling on dev (though is baked into the kernel so a bit pointless)

jrtc27, 2022-09-30:
Fixed by https://github.com/CTSRD-CHERI/cheribsd/pull/1361

closed 2022-09-30 by jrtc27