CTSRD-CHERI/cheribsd #1365: Unexpected mmap failure with MAP_ALIGNED and attempt to shadow (riscv64)

rmn30, 2022-05-06:
I am seeing an unexpected failure of mmap with the following test run on cheri-riscv64-purecap:
```
#include <stdio.h>
#include <sys/mman.h>
#include <assert.h>

int main (int argc, char **argv)
{
  void* p1 = mmap(0, 0x200000, PROT_READ | PROT_WRITE, MAP_PRIVATE | MAP_ANON | MAP_ALIGNED(21), -1, 0);
  assert(p1 != MAP_FAILED);
  void* p2;
  p2 = mmap(p1, 0x20000, PROT_READ | PROT_WRITE, MAP_PRIVATE | MAP_FIXED | MAP_ANON, -1, 0);
  assert(p2 == p1);
  p2 = mmap(p1, 0x40000, PROT_READ | PROT_WRITE, MAP_PRIVATE | MAP_FIXED | MAP_ANON, -1, 0);
  assert(p2 == p1);
  return 0;
}
```
The third mmap fails with ` errno 12 Cannot allocate memory`. Removing the `MAP_ALIGNED` from the first `mmap` or commenting out the second causes the third one to succeed.

For context, this was reduced from a `ktrace` of an `snmalloc` test. `snmalloc` uses `mmap` with `MAP_FIXED | MAP_ANON` to zero larger regions of memory after an initial reservation.

rmn30, 2022-05-09:
@bsdjhb , @nwf any thoughts on this?

brooksdavis, 2022-05-09:
I got started looking into this on Friday but had a lot of tooling problems. Thus far it looks like there's a failure to consolidate `MAP_ENTRY_UNMAPPED` entries after `vm_map_delete()` in `vm_map_fixed()`. This leads to two unmapped entries in a row from the same reservation, the first of which is 0x20000 long causing this to fail in `vm_map_insert()`:
```
                if ((new_entry->eflags & MAP_ENTRY_UNMAPPED) == 0 ||
                    new_entry->end < end)
                        return (KERN_NO_SPACE);
```
The problem seem to be that `vm_map_delete()` only attempts to consolidate downward and it needs to consolidate upward as well. I'm trying a build with the patch now.

brooksdavis, 2022-05-09:
My first patch didn't work, but this assertion detects the flaw I think is there:
```C
diff --git a/sys/vm/vm_map.c b/sys/vm/vm_map.c
index cdef2f8b0572..ea9fa01ff073 100644
--- a/sys/vm/vm_map.c
+++ b/sys/vm/vm_map.c
@@ -4439,6 +4439,13 @@ vm_map_delete(vm_map_t map, vm_offset_t start, vm_offset_t end,
                        }
                        /* Attempt to consolidate the mapping anyway? */
                        vm_map_try_merge_entries(map, scratch_entry, entry);
+                       /*
+                        * We should now have a single MAP_ENTRY_UNMAPPED
+                        * entry that at least spans whole delete
+                        * region.
+                        */
+                       KASSERT(entry->start <= start, ("entry->start (%ju) > start (%ju)", entry->start, start));
+                       KASSERT(entry->end >= end, ("entry->end (%ju) < end (%ju)", entry->end, end));
                } else {
                        vm_map_entry_delete(map, entry);
                }
```
The intuition is that now we've punched a hole in the existing reservation and there should be a single `MAP_ENTRY_UNMAPPED` entry spanning that space. Otherwise something has gone wrong.

qwattash, 2022-05-10:
Perhaps we should check for the assumed invariant that there are never two contiguous `VM_MAP_UNMAPPED` entries in `_vm_map_assert_consistent`? I'm currently trying to keep track of these design notes in a google doc. I have been meaning to add a section on coalescing policies.

brooksdavis, 2022-05-11:
Fixed in dev, will be merged to main soon.

commit 453cbe25 (2022-05-11): mmap: Fix MAP_FIXED over multiple entries

With reservations enabled, when a program performed an mmap MAP_FIXED
over a region with multiple VM map entries, the wrong previous entry
would sometimes be used when calling vm_map_try_merge_entries()
in vm_map_delete() for entries after the first one. This would
result in violating the invariant in vm_map_insert() that a single
MAP_ENTRY_UNMAPPED entry covers (at least) the entire region to map
into.

The invariant it general and MAP_ENTRY_UNMAPPED entries should always
be merged within a reservation.

Fixes #1365
closed 2022-05-11 by brooksdavis