CTSRD-CHERI/cheribsd #1379: panic when attempting NFS-root on Morello purecap kernel

nwf, 2022-05-16:
Well, things were looking great on *M*FS, and then we tried to pivot to *N*FS...

I don't think this is a re-emergence of https://github.com/CTSRD-CHERI/cheribsd/issues/213 but perhaps there's some analogous shenanigans in the in-kernel XDR code?

```
# reboot -r
Trying to mount root from nfs:10.190.106.73:/tank/morello/export/nwf-test-20220516 []...
  x0: 0xffffa0001b42d000 [rwxRW,0xffffa0001b42d000-0xffffa0001b42d400]
  x1: 0xffffa0001bbcc180 [rwxRW,0xffffa0001bbcc140-0xffffa0001bbcc180]
  x2: 0x0000000000000400
  x3: 0x0000000036313530
  x4: 0xffffa0001b42d040 [rwxRW,0xffffa0001b42d000-0xffffa0001b42d400]
  x5: 0xffffa0001bbcc140 [rwxRW,0xffffa0001bbcc140-0xffffa0001bbcc180]
  x6: 0xffff000000a29a65 [rR,0xffff000000a29a65-0xffff000000a29a74] (aarch32_esigcode + d2edd)
  x7: 0xffff000000a29a65 [rR,0xffff000000a29a65-0xffff000000a29a74] (aarch32_esigcode + d2edd)
  x8: 0x000000000000003c
  x9: 0x000000000000035a
 x10: 0x000000000004fb40
 x11: 0x00000000000001fc
 x12: 0x0000000000000000
 x13: 0x00000000af71e451
 x14: 0x00000000713a3b65
 x15: 0x00000000e0a2edb9
 x16: 0x0000000013fdcc9d
 x17: 0x0000000099b389ff
 x18: 0xffff0000ac024940 [rwxRW,0xffff0000ac022000-0xffff0000ac027000] (g_eli_key_cache_misses + a4006258)
 x19: 0xffff000000f25f90 [rwRW,0xffff000000f25f90-0xffff000000f26020] (M_TEMP + 0)
 x20: 0xffff00017e02d000 [rwxRW,0xffff00017e02d000-0xffff00017e02dc80]
 x21: 0xffff00017e02d150 [rwxRW,0xffff00017e02d000-0xffff00017e02dc80]
 x22: 0xffffa00004283000 [rwxRW,0xffffa00004283000-0xffffa00004283940]
 x23: 0x0000000000000002
 x24: 0x0000000000000000
 x25: 0xffffa0001b42d000 [rwxRW,0xffffa0001b42d000-0xffffa0001b42d400]
 x26: 0xffff0000ac024d00 [rwxRW,0xffff0000ac024d00-0xffff0000ac024d10] (g_eli_key_cache_misses + a4006618)
 x27: 0xffff0000ac024c90 [rwxRW,0xffff0000ac024c90-0xffff0000ac024d20] (g_eli_key_cache_misses + a40065a8)
 x28: 0xffff0000ac024c70 [rwxRW,0xffff0000ac022000-0xffff0000ac027000] (g_eli_key_cache_misses + a4006588)
 x29: 0xffff0000ac024940 [rwxRW,0xffff0000ac022000-0xffff0000ac027000] (g_eli_key_cache_misses + a4006258)
 ddc: 0x0000000000000000
  sp: 0xffff0000ac024940 [rwxRW,0xffff0000ac022000-0xffff0000ac027000]
  lr: 0xffff0000003cbe01 [rxR,0x0000000000000000-0xffffffffffffffff] (sentry) (nfs_mount + 1618)
 elr: 0xffff00000084cda8 [rxR,0x0000000000000000-0xffffffffffffffff] (_memcpy + 11f)
spsr:         24400045
 far: ffffa0001bbcc180
 esr:         9600002a
panic: Capability abort from kernel space: bounds violation
cpuid = 3
time = 1652729707
KDB: stack backtrace:
db_trace_self() at db_trace_self
db_trace_self_wrapper() at db_trace_self_wrapper+0x38
vpanic() at vpanic+0x1bc
panic() at panic+0x40
cap_abort() at cap_abort+0x250
handle_el1h_sync() at handle_el1h_sync+0x10
--- exception, esr 0x9600002a
_memcpy() at _memcpy+0x120
nfs_mount() at nfs_mount+0x1614
vfs_mount_sigdefer() at vfs_mount_sigdefer+0x2c
vfs_domount_first() at vfs_domount_first+0x248
vfs_domount() at vfs_domount+0x2a0
vfs_donmount() at vfs_donmount+0xa48
kernel_mount() at kernel_mount+0x54
parse_mount() at parse_mount+0x484
vfs_mountroot() at vfs_mountroot+0x928
sys_reboot() at sys_reboot+0x2dc
do_el0_sync() at do_el0_sync+0x648
handle_el0_sync() at handle_el0_sync+0x2c
--- exception, esr 0x56000000
KDB: enter: panic
[ thread pid 1 tid 100002 ]
Stopped at      kdb_enter+0x63: undefined       c200001f
```

jrtc27, 2022-05-16:
My guess is https://github.com/CTSRD-CHERI/cheribsd/blob/d870550ebe149e1002fe5eed2705c7dc76b73c11/sys/fs/nfsclient/nfs_clvfsops.c#L1298. That's definitely nonsense, though works when you don't have bounds. nfs_mount_parse_from just called `strdup(nam)` to give the result, so it copies a bunch of junk past the terminator (or everything but the terminator in the special case where nfs_mount_parse_from let it fit in nam's MNAMELEN+1 bytes but nfs_mount truncates it to MNAMELEN bytes).

nwf, 2022-05-17:
Well, I tried with
```
diff --git a/sys/fs/nfsclient/nfs_clvfsops.c b/sys/fs/nfsclient/nfs_clvfsops.c
index 829f866fff1e..10378d84c1eb 100644
--- a/sys/fs/nfsclient/nfs_clvfsops.c
+++ b/sys/fs/nfsclient/nfs_clvfsops.c
@@ -1295,7 +1295,7 @@ nfs_mount(struct mount *mp)
            &args.hostname, (struct sockaddr_in **)&nam, dirpath,
            sizeof(dirpath), &dirlen) == 0) {
                has_nfs_from_opt = 1;
-               bcopy(args.hostname, hst, MNAMELEN);
+               strncpy(hst, args.hostname, MNAMELEN);
                hst[MNAMELEN - 1] = '\0';

                /*
```
and it no longer panics, it just also doesn't work, hanging at:
```
# kenv vfs.root.mountfrom=nfs:10.190.106.73:/tank/morello/export/nwf-test-20220517
vfs.root.mountfrom="nfs:10.190.106.73:/tank/morello/export/nwf-test-20220517"
# reboot -r
Trying to mount root from nfs:10.190.106.73:/tank/morello/export/nwf-test-20220517 []...
```