CTSRD-CHERI/cheribsd #1459: freebsd64: sendmsg with control messages compatibility issue

brooksdavis, 2022-07-29:
Both freebsd32 and freebsd64 have the problem that control messages are expanded due to alignment padding when they are stuffed into an mbuf. Since they are limited to `MCLBYTES` (2048) in total, this means that a message that fits on a native system won't fit when run on a compatibility system. See this comment:
https://github.com/CTSRD-CHERI/cheribsd/blob/6a9e74b4bd73bef8cffdb0f134fc657fed60e029/sys/compat/freebsd32/freebsd32_misc.c#L1487-L1496

libnv works around this by always assuming 64-bit padding and we could chance it to always assume 128-bit padding, but that leaves existing aarch64 binaries stranded on Morello. 

See #1458 for a CheriABI workaround that could be expanded to hybrid or legacy.

bsdjhb, 2022-07-29:
Actually, there is a workaround for this, it just means using a non-normal mbuf cluster.  sendfile uses m_ext mbufs with an explicit external buffer with a custom free function that releases the vm_page_t.  In main now with ext_pgs an mbuf can now even span multiple pages, but basically the approach used in sendfile for the !ext_pgs case would work. Just malloc whatever size you need if it is larger than `MCLBYTES` and use a custom free routine that calls free with the right M_FOO.

I think the XXXMJ comment assumes one would have to use 4K clusters or the like, but that isn't necessarily required.

brooksdavis, 2022-10-20:
Fixed with the merge of #1461

closed 2022-10-20 by brooksdavis