CTSRD-CHERI/cheribsd #1464: CheriBSD/Morello elfdump raises capability fault

ianstark, 2022-08-08:
`/usr/bin/elfdump` on CheriBSD/Morello can terminate with a capability fault.
```
 $ elfdump -n a.out

note (.note.tag):
        FreeBSD 1400053 (NT_FREEBSD_ABI_TAG)
        CheriBSD 1593835520 (NT_FREEBSD_ABI_TAG)
In-address space security exception
```
It seems to be reliably generated by querying the note sections. Digging into gdb output suggests it's `elf_get_word` generating the fault, in a call from `elf_print_note`, probably this one at line 1135:
```
desc = elf_get_word(e, n + sizeof(Elf_Note) + namesz, 0);
```
That's in a loop, so clearly could be overstepping bounds, but I can't see why.

jrtc27, 2022-08-08:
This is because we don't round up namesz by 4 in all the pointer arithmetic; both name and desc are padded to 4 bytes, but `FreeBSD\0` is 8 bytes whilst `CheriBSD\0` is 9 bytes so we notice the problem downstream. This is also why the printed version is garbage, it's not skipping the padding, and `(20220511 & 0xff) << 24` is 1593835520. The undercounting then means it thinks there's another note. In general descsz also needs rounding up but our notes are 4-byte integers. This isn't true for other OSes though.

(Yes there is also a lack of any sanity checking here so actually-corrupt binaries can crash elfdump, but that's a separate issue...)

ianstark, 2022-08-08:
Ah, I did wonder about that surprise version number.