CTSRD-CHERI/cheribsd #1473: loader.efi doesn't like excess partitions?

nwf-msr, 2022-08-18:
I've set up a Morello board with an `/etc/fstab` of
```
# Device        Mountpoint      FStype  Options Dump    Pass#
/dev/ada0p3     /               ufs     rw      1       1
/dev/ada0p1     /boot/efi               msdosfs rw      2       2
/dev/ada0p4     /home           ufs     rw      2       2
/dev/ada0p2     none            swap    sw      0       0
```
and on boot I get told
```
Consoles: EFI console
    Reading loader env vars from /efi/freebsd/loader.env
Setting currdev to disk1p1:
FreeBSD/arm64 EFI loader, Revision 1.1
(Tue Jul  5 22:27:38 UTC 2022 jenkins@ctsrd-build-linux-xx)

   Command line arguments: loader.efi
   Image base: 0xf9359000
   EFI version: 2.70
   EFI Firmware: EDK II (rev 1.00)
   Console: efi (0x1000)
   Load Path: \EFI\BOOT\BOOTAA64.EFI
   Load Device: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x12,0x0)/Pci(0x0,0x
0)/Sata(0x0,0xFFFF,0x0)/HD(1,GPT,A49202E4-0835-11D4-A6CD-0002F70096D9,0x28,0x820
00)
   BootCurrent: 0005
   BootOrder: 0000 0001 0002 0003 0004 0005[*] 0006 0007
   BootInfo Path: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x12,0x0)/Pci(0x0,
0x0)/Sata(0x0,0xFFFF,0x0)
Ignoring Boot0005: Only one DP found
Trying ESP: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x12,0x0)/Pci(0x0,0x0)/S
ata(0x0,0xFFFF,0x0)/HD(1,GPT,A49202E4-0835-11D4-A6CD-0002F70096D9,0x28,0x82000)
Setting currdev to disk1p1:
Trying: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x12,0x0)/Pci(0x0,0x0)/Sata(
0x0,0xFFFF,0x0)/HD(2,GPT,BCC54CCF-0835-11D4-A6CD-0002F70096D9,0x82028,0x800000)
Setting currdev to disk1p2:
Trying: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x12,0x0)/Pci(0x0,0x0)/Sata(
0x0,0xFFFF,0x0)/HD(3,GPT,C89C5397-0835-11D4-A6CD-0002F70096D9,0x882028,0x3C00000
)
Setting currdev to disk1p3:
Trying: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x12,0x0)/Pci(0x0,0x0)/Sata(
0x0,0xFFFF,0x0)/HD(4,GPT,D0B29E16-0835-11D4-A6CD-0002F70096D9,0x4482028,0x17A000
00)
Setting currdev to disk1p4:
Failed to find bootable partition
ERROR: cannot open /boot/lua/loader.lua: no such file or directory.
```

If I run `set currdev=disk1p3:` and `boot` manually at the loader prompt, everything just works, so I'm not sure why the auto-probing sequence breaks but manual entry works.

I stuck some `printf`s in `libsa` to try to understand what was going on, but nothing informative emerged.

What else can I provide that would be useful?

jrtc27, 2022-08-18:
sanity_check_currdev is the thing to start with debugging I would imagine (stand/efi/loader/main.c).

nwf-msr, 2022-08-18:
Yeah, pushing through.  Let's see what we find with the power of violence`^W`_more `printf`_.

nwf-msr, 2022-08-18:
Some notes so far.  During the initial probe, we see things like...
```
Trying: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x12,0x0)/Pci(0x0,0x0)/Sata(
0x0,0xFFFF,0x0)/HD(3,GPT,C89C5397-0835-11D4-A6CD-0002F70096D9,0x882028,0x3C00000
)
NWF: that's a disk...
NWF: ... and a partition
Setting currdev to disk1p3:
NWF: efi_setcurrdev efi_parsedev(disk1p3:) -> 0
NWF: Trying open(disk1p3:/)
efipart_strategy disk blk=0 offset=0
efipart_strategy disk blk=64 offset=0
efipart_strategy disk blk=1 offset=0
efipart_strategy disk blk=1 offset=0
efipart_strategy disk blk=2 offset=0
efipart_strategy disk blk=468877311 offset=0
efipart_strategy disk blk=62914552 offset=8921128
efipart_strategy disk blk=0 offset=8921128
NWF: ... with dosfs
NWF: ufs_open 0 /
NWF: ufs_open 1
NWF: ufs_open read superblock
efipart_strategy disk blk=128 offset=8921128
efipart_strategy disk blk=40384 offset=8921128
NWF: ufs read_inode (s=0xf939f760 fdv=0xf3620d70)
efipart_strategy disk blk=320 offset=8921128
NWF: ufs_open post-strdup path
NWF: ufs_open out 0
NWF: ... with ufs
NWF: Trying open(disk1p3:/)
efipart_strategy disk blk=0 offset=0
efipart_strategy disk blk=64 offset=0
efipart_strategy disk blk=1 offset=0
efipart_strategy disk blk=1 offset=0
efipart_strategy disk blk=2 offset=0
efipart_strategy disk blk=468877311 offset=0
efipart_strategy disk blk=0 offset=8921128
NWF: ... with dosfs
NWF: ufs_open 0 /
NWF: ufs_open 1
NWF: ufs_open read superblock
efipart_strategy disk blk=128 offset=8921128
efipart_strategy disk blk=40384 offset=8921128
NWF: ufs read_inode (s=0xf939f760 fdv=0xf3620d70)
efipart_strategy disk blk=320 offset=8921128
NWF: ufs_open post-strdup path
NWF: ufs_open out 0
NWF: ... with ufs
NWF: efi_setcurrdev mount_currdev(2) -> 0
NWF: Trying open(/boot/defaults/loader.conf)
efipart_strategy disk blk=0 offset=0
efipart_strategy disk blk=64 offset=0
efipart_strategy disk blk=1 offset=0
efipart_strategy disk blk=1 offset=0
efipart_strategy disk blk=2 offset=0
efipart_strategy disk blk=468877311 offset=0
efipart_strategy disk blk=0 offset=532520
NWF: ... with dosfs
NWF: ufs_open 0 /boot/defaults/loader.conf
NWF: ufs_open 1
NWF: ufs_open read superblock
efipart_strategy disk blk=128 offset=532520
efipart_strategy disk blk=40384 offset=532520
NWF: ufs read_inode (s=0xf939f760 fdv=0xf3621340)
efipart_strategy disk blk=320 offset=532520
NWF: ufs_open post-strdup path
efipart_strategy disk blk=106176 offset=532520
NWF: ufs read_inode (s=0xf939f760 fdv=0xf3621340)
efipart_strategy disk blk=265053056 offset=532520
NWF: ufs read_inode read strat fail
```
and if manually set we see
```
OK set currdev=disk1p3:
NWF: efi_setcurrdev efi_parsedev(disk1p3:) -> 0
NWF: efi_setcurrdev mount_currdev(2) -> 0
OK ls /boot/defaults/loader.conf
NWF: Trying open(/boot/defaults/loader.conf)
efipart_strategy disk blk=0 offset=0
efipart_strategy disk blk=64 offset=0
efipart_strategy disk blk=1 offset=0
efipart_strategy disk blk=1 offset=0
efipart_strategy disk blk=2 offset=0
efipart_strategy disk blk=468877311 offset=0
efipart_strategy disk blk=0 offset=8921128
NWF: ... with dosfs
NWF: ufs_open 0 /boot/defaults/loader.conf
NWF: ufs_open 1
NWF: ufs_open already mounted
NWF: ufs read_inode (s=0xf939f760 fdv=0xf3621510)
efipart_strategy disk blk=320 offset=8921128
NWF: ufs_open post-strdup path
efipart_strategy disk blk=106176 offset=8921128
NWF: ufs read_inode (s=0xf939f760 fdv=0xf3621510)
efipart_strategy disk blk=23048384 offset=8921128
efipart_strategy disk blk=23088448 offset=8921128
NWF: ufs read_inode (s=0xf939f760 fdv=0xf3621510)
efipart_strategy disk blk=23048384 offset=8921128
efipart_strategy disk blk=23088496 offset=8921128
NWF: ufs read_inode (s=0xf939f760 fdv=0xf3621510)
efipart_strategy disk blk=23048384 offset=8921128
NWF: ufs_open out 0
NWF: ... with ufs
    /boot/defaults/loader.conf
```

The most important lines so far are the contrast between
```
efipart_strategy disk blk=265053056 offset=532520
```
and the correct
```
efipart_strategy disk blk=23048384 offset=8921128
```

The former appears to be gibberish!  Importantly, the `offset` value there is not the offset of `disk1p3:` in the disk but rather the offset of `disk1p2:`!
```
=>       40  468877232  ada0  GPT  (224G)
         40     532480     1  efi  (260M)
     532520    8388608     2  freebsd-swap  (4.0G)
    8921128   62914560     3  freebsd-ufs  (30G)
   71835688  396361728     4  freebsd-ufs  (189G)
  468197416     679856        - free -  (332M)
```

So somehow when probing we get "stuck" on the wrong partition, even though we used the right one when mounting.  WTF.

nwf-msr, 2022-08-18:
Spidey senses tingling.... there _was_ a UFS filesystem at that offset on that disk, and it probably wasn't zeroed despite having a swap partition land on it instead...

Yup!  Zeroing that partition resulted in the machine automatically booting the right partition.

jrtc27, 2022-08-19:
Can you reproduce the same thing with FreeBSD head?

jrtc27, 2022-08-19:
(You can even just nab loader, we haven't modified it in substantive ways)

nwf, 2022-08-23:
Well, FWIW, I just grabbed an amd64 installer for 13.1 and replayed the installations, first placing `/` before `swap` and then again reversing them and the system boots fine... with the non-EFI-based loader.  D'oh.

Doing that _again again_ with EFI, and this time, the system exhibits the bug.  So it's not our fault, and it looks to be EFI-specific.