CTSRD-CHERI/cheribsd #1502: Capability relocation for non-cheri binary

bstellaceleste, 2022-10-04:
When compilling WAVM (https://github.com/WAVM/WAVM - that does not have support for freebsd) on the Morello machine, I'm unable to execute the generated binary and llvm-objdump on the libWAVM.so lib gives this output

```
root@cheribsd:/mnt/WAVM/build_dir # llvm-objdump -dRsg  libWAVM.so | head

libWAVM.so:     file format elf64-littleaarch64
DYNAMIC RELOCATION RECORDS
0000000003afc978 R_MORELLO_RELATIVE *ABS*+0x19ca7e4
0000000003afc980 R_MORELLO_RELATIVE *ABS*+0x1a1f10c
0000000003afc988 R_MORELLO_RELATIVE *ABS*+0x1c75610
0000000003afc990 R_MORELLO_RELATIVE *ABS*+0x1c826fc
0000000003afc998 R_MORELLO_RELATIVE *ABS*+0x1d11f58
0000000003afc9a0 R_MORELLO_RELATIVE *ABS*+0x1d3052c
0000000003afc9a8 R_MORELLO_RELATIVE *ABS*+0x1d3df88
```

jrtc27, 2022-10-04:
As discussed on Slack this is what happens when you mix hybrid and purecap code, which is invalid. Newer Morello LLVM versions than in 22.05 can catch some of those cases but not all of them.

closed 2022-10-04 by jrtc27

bstellaceleste, 2022-10-04:
I still don't understand;; WAVM does not mix hybrid and purecap mode

jrtc27, 2022-10-04:
Perhaps not deliberately, but somehow you’re ending up with that.