CTSRD-CHERI/cheribsd #1528: Valid capabilities can be read from files in a tmpfs

gabi-250, 2022-10-21:
According to the CHERI C/C++ programming guide, it shouldn't be possible to read a valid capability from a file:

> In the CheriABI process environment, only untagged data (not tagged pointers) may be written to or read from files. If a region of memory containing valid pointers is written to a file, and then read back, the pointers in that region will no longer be valid.

 However, as the example below shows, one _can_ read a valid capability from a file if the file is on a tmpfs:
 ```c
/*
 * This example writes a capability to the file specified in argv[1], and then
 * attempts to read it back. It will read back a valid capability if argv[1]
 * specifies a path on a tmpfs.
 *
 */
#include <cheriintrin.h>

#include <assert.h>
#include <cheri/cheric.h>
#include <errno.h>
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>

#define CAP_SIZE sizeof(uintptr_t)

void
write_cap(FILE *f, uintptr_t *cap)
{
    size_t n = write(fileno(f), cap, CAP_SIZE);
    // Assume we can write `CAP_SIZE` bytes in one go.
    if (n != CAP_SIZE) {
        fprintf(stderr, "write failed: %zu\n", n);
        exit(1);
    }

    if (fflush(f)) {
        perror("failed to flush");
        exit(1);
    }
}

void
read_cap(FILE *f, uintptr_t *cap)
{
    if (fseek(f, 0, SEEK_SET)) {
        perror("failed to seek");
        exit(1);
    }

    size_t n = read(fileno(f), cap, CAP_SIZE);
    // Assume there are `CAP_SIZE` bytes available to read.
    if (n != CAP_SIZE) {
        fprintf(stderr, "read failed: %zu\n", n);
        exit(1);
    }
}

int
main(int argc, char **argv)
{
    int x = 1;
    // The capability that will be written to the file specified in argv[1]
    uintptr_t cap = (uintptr_t)&x;
    assert(cheri_is_valid((void *)cap));

    if (argc != 2) {
        fprintf(stderr, "invalid number of arguments: expected 1, found %d\n", argc - 1);
        return 1;
    }

    FILE *f = NULL;
    if (!(f = fopen(argv[1], "a+"))) {
        perror("failed to open file");
        return 1;
    }

    printf("Writing capability to file: %#lp\n", (void *)cap);
    // Write the capability to the specified path
    write_cap(f, &cap);

    uintptr_t stored_cap = 0;
    // Sanity check: 0 is not a valid capability
    assert(!cheri_is_valid((void *)stored_cap));
    // Read back the capability written earlier
    read_cap(f, &stored_cap);

    printf("Read capability from file: %#lp\n", (void *)stored_cap);
    assert(cap == stored_cap);
    // XXX the capability should not be valid but due to a bug in CheriBSD
    // this assertion succeeds!
    assert(cheri_is_valid((void *)stored_cap));
    return 0;
}
 ```

 On a system where `/tmp` is mounted as tmpfs and `~` on a UFS file system, we get the following output:

```
root@cheribsd-morello-purecap:~ # ./cap_to_file /tmp/cap.bin
Writing capability to file: 0xfffffff7ff5c [rwRW,0xfffffff7ff5c-0xfffffff7ff60]
Read capability from file: 0xfffffff7ff5c [rwRW,0xfffffff7ff5c-0xfffffff7ff60]
root@cheribsd-morello-purecap:~ # ./cap_to_file ~/cap.bin
Writing capability to file: 0xfffffff7ff5c [rwRW,0xfffffff7ff5c-0xfffffff7ff60]
Read capability from file: 0xfffffff7ff5c [rwRW,0xfffffff7ff5c-0xfffffff7ff60] (invalid)
Assertion failed: (cheri_is_valid((void *)stored_cap)), function main, file cap_to_file.c, line 87.
Abort trap (core dumped)
root@cheribsd-morello-purecap:~ #
```

This bug can also be exploited to "leak" a capability from one process to another (if a process writes a capability to a file in a tmpfs, a different process can later read the file to retrieve a valid capability - see [leak-capability]).

I could reproduce this issue on CheriBSD (purecap mode) in QEMU, FVP, as well as on a Morello board.

I _think_ this happens because [`tmpfs_read`] and [`tmpfs_write`] use [`uiomove_object`], which eventually calls [`uiomove_from_phys_cap`] (which calls `uiomove_fromphys_flags` with `preserve_flags = true`).

[leak-capability]: https://github.com/capablevms/cheri-examples/tree/359e351837d8e162798f88d32554999d4423be68/capability_sharing/leak-capability
[`tmpfs_read`]: https://github.com/CTSRD-CHERI/cheribsd/blob/4511a9b9d97ab3ca8041873e353d818d1541d8fc/sys/fs/tmpfs/tmpfs_vnops.c#L596
[`tmpfs_write`]: https://github.com/CTSRD-CHERI/cheribsd/blob/4511a9b9d97ab3ca8041873e353d818d1541d8fc/sys/fs/tmpfs/tmpfs_vnops.c#L677
[`uiomove_object`]: https://github.com/CTSRD-CHERI/cheribsd/blob/4511a9b9d97ab3ca8041873e353d818d1541d8fc/sys/kern/uipc_shm.c#L247
[`uiomove_from_phys_cap`]: https://github.com/CTSRD-CHERI/cheribsd/blob/4511a9b9d97ab3ca8041873e353d818d1541d8fc/sys/arm64/arm64/uio_machdep.c#L161

cc @ltratt @jacobbramley

bsdjhb, 2022-10-21:
Probably `uiomove_object_page` needs to be honoring `OBJ_HASCAP` and using that flag to decide whether or not to preserve capabilities.  However, I think I might like to refactor how we handle capabilities with the uiomove(9) API to make this simpler to do.

rwatson, 2022-10-21:
FWIW, I agree that this is a bug. While the tags in files is an exciting potential research area .. it’s not the way we currently envision CheriBSD / POSIX working until that research is done and proves to be a good idea :-).

gabi-250, 2022-10-21:
Nice, thanks for the quick turnaround!

> FWIW, I agree that this is a bug. While the tags in files is an exciting potential research area .. it’s not the way we currently envision CheriBSD / POSIX working until that research is done and proves to be a good idea :-).

Ha, I agree - I wasn't expecting it to be a valid use-case, I just happened to stumble upon this bug while testing some corner cases :)

bsdjhb, 2022-11-29:
This was fixed in the linked PR.

commit 2b56450a (2022-11-02): uiomove_object_page: Do not preserve tags.

This routine is only used for read(2) and write(2) of file descriptors
backed by VM objects.  However, read(2) and write(2) are not really
sensible APIs for preserving tags.  We could perhaps use OBJ_HASCAP on
the VM object to decide if we should preserve tags for read/write of
shm_open(2) file descriptors, but that use case doesn't seem useful.

Closes: #1528
closed 2022-11-29 by bsdjhb